dtcooper/raspotify #652: Raspotify restart right after I connect

shumafuk, 2024-02-23:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Raspotify suddenly stop playing anything. It disconnect right after I connect with any client at network, restart and after while I can connect again but again it disconnect, restart... so on. Any help from logs are appreciated. Thanks.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
my@rpi:~ $ sudo journalctl -u raspotify -b
-- Journal begins at Wed 2023-05-03 05:11:24 CEST, ends at Fri 2024-02-23 10:03:36 CET. --
Feb 22 09:17:07 rpi systemd[1]: Started Raspotify (Spotify Connect Client).
Feb 22 09:17:18 rpi librespot[479]: [2024-02-22T08:17:18Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Feb 22 09:17:18 rpi librespot[479]: [2024-02-22T08:17:18Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Feb 22 09:17:18 rpi librespot[479]: [2024-02-22T08:17:18Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Feb 22 09:17:18 rpi librespot[479]: [2024-02-22T08:17:18Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:4070
Feb 23 09:01:19 rpi librespot[479]: [2024-02-23T08:01:19Z WARN  librespot_connect::spirc] No autoplay_uri found
Feb 23 09:01:19 rpi librespot[479]: [2024-02-23T08:01:19Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Feb 23 09:01:20 rpi librespot[479]: [2024-02-23T08:01:20Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_ope>
Feb 23 09:01:20 rpi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Feb 23 09:01:20 rpi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Feb 23 09:01:20 rpi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Feb 23 09:01:20 rpi systemd[1]: raspotify.service: Consumed 2.356s CPU time.
Feb 23 09:01:32 rpi systemd[1]: Started Raspotify (Spotify Connect Client).
Feb 23 09:01:33 rpi librespot[2056]: [2024-02-23T08:01:33Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Feb 23 09:01:33 rpi librespot[2056]: [2024-02-23T08:01:33Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Feb 23 09:01:33 rpi librespot[2056]: [2024-02-23T08:01:33Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Feb 23 09:01:33 rpi librespot[2056]: [2024-02-23T08:01:33Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:4070
Feb 23 09:01:44 rpi librespot[2056]: [2024-02-23T08:01:44Z WARN  librespot_connect::spirc] No autoplay_uri found
Feb 23 09:01:44 rpi librespot[2056]: [2024-02-23T08:01:44Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Feb 23 09:01:45 rpi librespot[2056]: [2024-02-23T08:01:45Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_op>
Feb 23 09:01:45 rpi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Feb 23 09:01:45 rpi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Feb 23 09:01:45 rpi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Feb 23 09:01:56 rpi systemd[1]: Started Raspotify (Spotify Connect Client).
Feb 23 09:01:57 rpi librespot[2504]: [2024-02-23T08:01:57Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Feb 23 09:01:57 rpi librespot[2504]: [2024-02-23T08:01:57Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Feb 23 09:01:57 rpi librespot[2504]: [2024-02-23T08:01:57Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Feb 23 09:42:13 rpi librespot[2504]: [2024-02-23T08:42:13Z WARN  librespot_connect::spirc] No autoplay_uri found
Feb 23 09:42:13 rpi librespot[2504]: [2024-02-23T08:42:13Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Feb 23 09:42:16 rpi librespot[2504]: [2024-02-23T08:42:16Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_op>
Feb 23 09:42:16 rpi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Feb 23 09:42:16 rpi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Feb 23 09:42:16 rpi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
[... 244 lines cut ...]

                Steinberg UR22mkII, USB Audio
                IEC958 (S/PDIF) Digital Audio Output

        Supported Format & Sample Rate Combinations:

                Format: S32   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz

        --------------------------------------------------------------------

        Device:

                dmix:CARD=UR22mkII,DEV=0

        Description:

                Steinberg UR22mkII, USB Audio
                Direct sample mixing device

        Supported Format & Sample Rate Combinations:

                Format: S32   Sample Rate(s): 48kHz

        --------------------------------------------------------------------
```

shumafuk, 2024-02-26:
Somehow its working now, not sure why. Problem solved.

DatenDynamo, 2024-03-03:
I had this problem in the past with another client. The problem with that one was that an image output had to be connected. Unfortunately, it didn't work with raspotify to simply connect a monitor via HDMI

leonnicolas, 2024-05-01:
I also had the ERROR
> ` Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_op>`

Could fix it by defining the device (`LIBRESPOT_DEVICE`) name in `/etc/raspotify/conf` to an actual device, not default. I could list possible devices with `librespot -d ?`. Maybe you will also need to define the format (`LIBRESPOT_FORMAT`) in the same file.

So annoying, because simply running `librespot` in a ssh session works without defining the exact device, not when starting it with systemd though.

arthurmloureiro, 2024-05-04:
> I also had the ERROR
>
>
> Could fix it by defining the device (`LIBRESPOT_DEVICE`) name in `/etc/raspotify/conf` to an actual device, not default. I could list possible devices with `librespot -d ?`. [...]

I am also having this issue... can you explain better how you solved it?

leonnicolas, 2024-05-05:
For example on my raspberry pi 4: I type 
```shell
librespot -d ?
```
to find possible alsa devices. In my case the first one is called `hw:CARD=Headphones,DEV=0`.

I open the file `/etc/raspotify/conf` and uncomment the entry `LIBRESPOT_DEVICE` and set its value to `hw:CARD=Headphones,DEV=0`.

I also had to set the format to `S16`. Maybe for your pi this can be another format. The format can be defined in `/etc/raspotify/conf` as well. The entry is called `LIBRESPOT_FORMAT`.

I restart the systemd unit. 
```shell 
systemctl restart raspotify
```
Check the logs with 
```shell 
journalctl -fu raspotify
```
while trying to play a song. Hopefully it works.

theemas, 2024-05-28:
for me it ended up giving this code when listing `librespot -d ?` 
pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave 

i had to disable the on board audio to only get the installed Dac when listing `librespot -d ?`
after that it worked flawlessly

closed 2024-09-03 by github-actions[bot]