dtcooper/raspotify #655: What is the purpose of the librespot fork

XuluWarrior, 2024-03-17:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Trying to debug an issue with raspotify (which magically fixed itself the next day) I noticed that although [librespot-org/librespot](https://github.com/librespot-org/librespot) continues to be relatively actively maintained, [JasonLG1979/librespot](https://github.com/JasonLG1979/librespot) is less so.
In fact although the fork is 114 commits ahead it is also 376 commits behind.

What is the purpose of the fork?  Does it have changes that haven't/can't be merged upstream?
If maintaining the fork has a cost wouldn't it be better to wrap librespot-org/librespot directly?

XuluWarrior, 2024-06-18:
I'm pinging this to keep my question alive a bit longer as the bot wants to close this due to inactivity.

If anybody knows why the librespot fork exists that would be appreciated.  Otherwise, I'll let this issue die.

l3d00m, 2024-08-15:
I remember that @JasonLG1979 started the fork because he was annoyed with some decisions in the OG librespot. I'm distinctively remember there being an issue or discussion in this repo or in the librespot repo, or at least in the commit names. 

But yes, as you've rightly pointed out, it just made it harder to find out where to file issues and now there's just an additional abandonware which makes it harder to take over the maintenance of raspotify. Especially I've found the original librespot authors quite friendly and responsive and this maintainer now seems MIA.

maxenced, 2024-08-21:
Also, just tested it, the main librespot repository works as a drop-in replacement with the fork. So no point in keeping it for me

sowthistle, 2024-09-03:
@maxenced 

> Also, just tested it, the main librespot repository works as a drop-in replacement with the fork. So no point in keeping it for me

I'm entirely new to raspotify and it seems I've arrived at the party just as some librespot-related thing has broken (#669). If the raspotify fork of librespot is not being actively maintained, it seems I might be better off using the main version instead (in the hope that it gets fixed soon). Could you say what is involved in using it as a drop-in?

kimtore, 2024-09-17:
Hello @sowthistle 

Take a look at #672 which reverts to using upstream on the dev branch.

I hope to see this merged soon so we can all try out the new version, there have been quite a bit of change since that fork was last updated.

l3d00m, 2024-09-17:
@kimtore Thanks a lot for bringing librespot back to life!!