dtcooper/raspotify #662: Raspotify won't show up in device list, even through it is running

statewarrior483, 2024-06-18:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Raspberry Pi 4B

So pretty much I had the whole thing working PERFECTLY 2 days ago, but then I shut down my rpi and I went out for the weekend with my family. After I returned, I found tht even when I start it up, I dont see it in the device list. I even do `sudo systemctl status raspotify` and it says its active, I tried restarting it using `sudo systemctl restart raspotify` and again, no fix. I tried COMPLETELY REINSTALLING RASPBERRY OS. Nothing happened. I am using raspberryOS bullseye 32-bit (because it only works on here and I am using a youtube inspiration https://www.youtube.com/watch?v=-jGWjFR936o ).

```
sudo systemctl status raspotify

● raspotify.service - Raspotify (Spotify Connect Client)

Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor pre>

Active: active (running) since Mon 2024-06-17 17:06:42 CDT; 1s ago

Docs: https://github.com/dtcooper/raspotify

https://github.com/librespot-org/librespot

https://github.com/dtcooper/raspotify/wiki

https://github.com/librespot-org/librespot/wiki/Options

Main PID: 4288 (librespot)

Tasks: 1 (limit: 4915)

CPU: 118ms

CGroup: /system.slice/raspotify.service

└─4288 /usr/bin/librespot

Jun 17 17:06:42 raspberrypi systemd[1]: Started Raspotify
```

This is the output of checking the status.

I even did a spotify developer API call ( https://developer.spotify.com/documentation/web-api/reference/get-a-users-available-devices ) and again, the pi did not pop up.

This is odd because it worked fine 2 days ago. And yes, I am on the same network as the pi.

Please help.

statewarrior483, 2024-06-18:
Update:
This is the output when I checked the log file:
`sudo journalctl -u raspotify.service
-- Journal begins at Mon 2024-03-11 20:20:38 CDT, ends at Mon 2024-06-17 22:46:>
Jun 17 16:57:24 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Clie>
Jun 17 17:06:42 raspberrypi systemd[1]: Stopping Raspotify (Spotify Connect Cli>
Jun 17 17:06:42 raspberrypi systemd[1]: raspotify.service: Succeeded.
Jun 17 17:06:42 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Clie>
Jun 17 17:06:42 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Clie>
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdn>
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdn>
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdn>
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdn>
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdn>
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdn>
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdn>
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdn>
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdn>
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdn>
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdn>
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdn>
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdn>
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdn>
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdn>
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdn>
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdn>
lines 1-23...skipping...
-- Journal begins at Mon 2024-03-11 20:20:38 CDT, ends at Mon 2024-06-17 22:46:13 CDT. --
Jun 17 16:57:24 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Jun 17 17:06:42 raspberrypi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Jun 17 17:06:42 raspberrypi systemd[1]: raspotify.service: Succeeded.
Jun 17 17:06:42 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Jun 17 17:06:42 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:40 raspberrypi librespot[4288]: [2024-06-17T22:22:40Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:41 raspberrypi librespot[4288]: [2024-06-17T22:22:41Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:43 raspberrypi librespot[4288]: [2024-06-17T22:22:43Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 17:22:47 raspberrypi librespot[4288]: [2024-06-17T22:22:47Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 17:22:55 raspberrypi librespot[4288]: [2024-06-17T22:22:55Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
[... cut ...]
Jun 17 19:22:41 raspberrypi librespot[4288]: [2024-06-18T00:22:41Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 19:22:41 raspberrypi librespot[4288]: [2024-06-18T00:22:41Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
Jun 17 19:22:43 raspberrypi librespot[4288]: [2024-06-18T00:22:43Z WARN  libmdns::fsm] couldn't parse packet from 192.168.68.85:5353: type 29 is invalid
Jun 17 19:22:43 raspberrypi librespot[4288]: [2024-06-18T00:22:43Z WARN  libmdns::fsm] couldn't parse packet from [fe80::221:b7ff:fed7:f97b%3]:5353: type 29 is invalid
`

statewarrior483, 2024-06-18:
Avahi log file:
`-- Journal begins at Mon 2024-03-11 20:20:38 CDT, ends at Mon 2024-06-17 22:46:>
Mar 11 20:20:42 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: Found user 'avahi' (UID 108) and>
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: Successfully dropped root privil>
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: avahi-daemon 0.8 starting up.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Successfully called chroot().
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Successfully dropped remaining c>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: No service file found in /etc/av>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on >
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: New relevant interface lo.IPv6 f>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on >
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: New relevant interface lo.IPv4 f>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Network interface enumeration co>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Registering new address record f>
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Registering new address record f>
Mar 11 20:20:43 raspberrypi systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Mar 11 20:20:44 raspberrypi avahi-daemon[393]: Server startup complete. Host na>
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on >
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: New relevant interface wlan0.IPv>
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: Registering new address record f>
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Leaving mDNS multicast group on >
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on >
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Registering new address record f>
lines 1-23...skipping...
-- Journal begins at Mon 2024-03-11 20:20:38 CDT, ends at Mon 2024-06-17 22:46:13 CDT. --
Mar 11 20:20:42 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: Successfully dropped root privileges.
Mar 11 20:20:42 raspberrypi avahi-daemon[393]: avahi-daemon 0.8 starting up.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Successfully called chroot().
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Successfully dropped remaining capabilities.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: No service file found in /etc/avahi/services.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: New relevant interface lo.IPv6 for mDNS.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: New relevant interface lo.IPv4 for mDNS.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Network interface enumeration completed.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Registering new address record for ::1 on lo.*.
Mar 11 20:20:43 raspberrypi avahi-daemon[393]: Registering new address record for 127.0.0.1 on lo.IPv4.
Mar 11 20:20:43 raspberrypi systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Mar 11 20:20:44 raspberrypi avahi-daemon[393]: Server startup complete. Host name is raspberrypi.local. Local service cookie is 2296560523.
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::b560:72ae:e4ea:5488.
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: New relevant interface wlan0.IPv6 for mDNS.
Mar 11 20:22:39 raspberrypi avahi-daemon[393]: Registering new address record for fe80::b560:72ae:e4ea:5488 on wlan0.*.
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::b560:72ae:e4ea:5488.
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on interface wlan0.IPv6 with address ::1017.
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Registering new address record for ::1017 on wlan0.*.
Mar 11 20:22:41 raspberrypi avahi-daemon[393]: Withdrawing address record for fe80::b560:72ae:e4ea:5488 on wlan0.
Mar 11 20:22:43 raspberrypi avahi-daemon[393]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.68.83.
Mar 11 20:22:43 raspberrypi avahi-daemon[393]: New relevant interface wlan0.IPv4 for mDNS.
Mar 11 20:22:43 raspberrypi avahi-daemon[393]: Registering new address record for 192.168.68.83 on wlan0.IPv4.
Mar 11 20:22:44 raspberrypi avahi-daemon[393]: Got SIGHUP, reloading.
Mar 11 20:22:44 raspberrypi avahi-daemon[393]: No service file found in /etc/avahi/services.
Mar 11 20:22:46 raspberrypi avahi-daemon[393]: Got SIGHUP, reloading.
Mar 11 20:22:46 raspberrypi avahi-daemon[393]: No service file found in /etc/avahi/services.
-- Boot e0aad4f1e3b24b2c9dc79163e8f6e8b7 --
Jun 17 16:41:20 raspberrypi systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Jun 17 16:41:21 raspberrypi avahi-daemon[395]: Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Jun 17 16:41:21 raspberrypi avahi-daemon[395]: Successfully dropped root privileges.
Jun 17 16:41:21 raspberrypi avahi-daemon[395]: avahi-daemon 0.8 starting up.
Jun 17 16:41:21 raspberrypi avahi-daemon[395]: Successfully called chroot().
[... cut ...]
Jun 17 16:42:24 raspberrypi avahi-daemon[405]: Server startup complete. Host name is raspberrypi.local. Local service cookie is 3590802785.
Jun 17 16:42:31 raspberrypi avahi-daemon[405]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::b560:72ae:e4ea:5488.
Jun 17 16:42:31 raspberrypi avahi-daemon[405]: New relevant interface wlan0.IPv6 for mDNS.
Jun 17 16:42:31 raspberrypi avahi-daemon[405]: Registering new address record for fe80::b560:72ae:e4ea:5488 on wlan0.*.
lines 1-82
`

statewarrior483, 2024-06-18:
Here are the gdocs for the logs, just to make it easier to read ig

Raspotify logs: https://docs.google.com/document/d/1GhxHYL4kxib-BaTR5vkxBj_Ty0uOB602mg87UoDXa_0/edit?usp=sharing

Avahi logs: https://docs.google.com/document/d/1tykQencchIvzkW7mH54HxgJSb-ZNpzdCMH8IslOzKeo/edit?usp=sharing

closed 2024-09-24 by github-actions[bot]

kimtore, 2024-09-24:
@statewarrior483, please try again with the newest release 0.44.1?