dtcooper/raspotify #663: Raspotify only shows in device list via API after Spotify App selects it

optikalefx, 2024-06-19:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

When using the Spotify Web API, Raspotify is not in the device list via API.

```python
deviceResponse = spotify.sp.devices()
    find_ras = lambda v: "raspotify" in v['name']
    raspotify = next(filter(find_ras, deviceResponse['devices']), None)
    return bool(raspotify)
```

However, it IS showing in the spotify mac app

[image]

Once I choose Raspotify in the Spotify Mac App, then the Spotify Web Api will return the device and can be used.

Do we have any idea why the Mac App has to first choose Raspotify before it can be a visible device in the API? I've made the API calls using Postman, to rule out any possible issue with Spotipy. 

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
sudo journalctl -u raspotify -b
Jun 19 01:54:37 raspberrypi systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
```

kimtore, 2024-09-18:
@optikalefx,

Please try the latest release and see if it fixes your problem.

mitea1, 2024-09-23:
Same issue here. Also on version 0.4.0

kimtore, 2024-09-26:
Not sure what causes this, but I would check if mDNS is working properly. Is `avahi-daemon` installed and enabled? Are your devices on the same network subnet, and if not have you enabled mDNS repeater?

mitea1, 2024-09-26:
I get this on my rpi:

```
pi@raspberrypi:~ $ sudo systemctl status avahi-daemon
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
     Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-09-23 15:57:43 CEST; 3 days ago
TriggeredBy: ● avahi-daemon.socket
   Main PID: 546 (avahi-daemon)
     Status: "avahi-daemon 0.8 starting up."
      Tasks: 2 (limit: 755)
        CPU: 384ms
     CGroup: /system.slice/avahi-daemon.service
             ├─546 "avahi-daemon: running [raspberrypi-2.local]"
             └─551 "avahi-daemon: chroot helper"

Sep 23 15:57:48 raspberrypi avahi-daemon[546]: Withdrawing address record for (obfuscated) on wlan0.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Withdrawing address record for 192.168.178.26 on wlan0.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Withdrawing address record for ::1 on lo.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Withdrawing address record for 127.0.0.1 on lo.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Host name conflict, retrying with raspberrypi-2
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Registering new address record for (obfuscated) on wlan0.*.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Registering new address record for 192.168.178.26 on wlan0.IPv4.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Registering new address record for ::1 on lo.*.
Sep 26 20:35:41 raspberrypi avahi-daemon[546]: Registering new address record for 127.0.0.1 on lo.IPv4.
Sep 26 20:35:42 raspberrypi avahi-daemon[546]: Server startup complete. Host name is raspberrypi-2.local. Local service cookie is 4003845144.
```
dont know where the first hostname conflict comes from and if it has something to do with my problem.

I connect my rpi with raspotify to the same wifi as my laptop. If I use the spotify app on my laptop i see the raspotify device and can select it. However the flutter app running on the same rpi doesn't get the device listed via spotify api request initially. Only after I first selected it from my laptop then its possible for the flutter app to get it listed via spotify api request.

I will test if I get the device from my laptop if I use postman and use the same spotify api request.

kimtore, 2024-09-27:
To the best of my knowledge, there are two ways that Spotify clients may see available devices:
- through the Spotify Web API, where you will get devices which have recently authenticated with Spotify
- through mDNS and the Spotify Connect TCP service in librespot (must be on the same network, not be blocked in the firewall, or with mDNS repeater)

You can check if the service is discoverable by running:
```
$ avahi-browse -a -t | grep spotify
+   eth0 IPv6 Anlegget                                      _spotify-connect._tcp local
+   eth0 IPv4 Anlegget                                      _spotify-connect._tcp local
```

mitea1, 2024-09-27:
> avahi-browse -a -t | grep spotify

I get 

```
pi@rpi:~ $ avahi-browse -a -t | grep spotify
+  wlan0 IPv6 raspotify (rpi)                               _spotify-connect._tcp local
+  wlan0 IPv4 raspotify (rpi)                               _spotify-connect._tcp local

```

I still dont get why the laptop in the same network is able to "see" it when using the official app but not when using the web api. And why it needs this initial selection. I dont know the details but my guess is that the official app is either using a different API and/or somehow authorizing the raspotify device when it selects it.

kimtore, 2024-09-30:
> I still dont get why the laptop in the same network is able to "see" it when using the official app but not when using the web api. And why it needs this initial selection.
This is my understanding of how it works:

Spotify's Web API doesn't know about your device because yet because the device is not *logged in* to Spotify with your user credentials.

The official app uses auto-discovery (through avahi) to find devices on the local network, and merges the list of local devices with the list of devices it got from the Web API. However, when you connect to a device, the device info gets sent to Spotify servers, and during the time that device is "active" you can also access it through the Web API.

mitea1, 2024-10-01:
> . However, when you connect to a device, the device info gets sent to Spotify servers, and during the time that device is "active" you can also access it through the Web API

So if I understand correctly, a device is not visible (recognized by the Spotify servers) before this step. Only the official app or the device itself, when it logs in, can add the device to the servers.

I assume the device can only log in using the credentials defined [here](https://www.spotify.com/us/account/set-device-password/)? These would be the credentials LIBRESPOT_USERNAME and LIBRESPOT_PASSWORD that need to be added to the etc/raspotify/conf file, correct? However, I always get an error stating that the password couldn't be set when I try to create one.

Which brings me to the next point: Unfortunately, it seems like Spotify has disabled the option to use credentials. There are other issues related to this (https://github.com/Spotifyd/spotifyd/issues/1293) where they mention that this might be the case.

Do you have any details or can confirm that?

kimtore, 2024-10-15:
> So if I understand correctly, a device is not visible (recognized by the Spotify servers) before this step.
I believe so.

> Only the official app or the device itself, when it logs in, can add the device to the servers.
Also third party apps will do this.

> Which brings me to the next point: Unfortunately, it seems like Spotify has disabled the option to use credentials. There are other issues related to this ([Spotifyd/spotifyd#1293](https://github.com/Spotifyd/spotifyd/issues/1293)) where they mention that this might be the case.
> 
> Do you know any details and/or can confirm that?

See #675 for the credentials issue.

kimtore, 2024-10-18:
I'm going to close this issue, feel free to comment and/or re-open if you have any other questions.

closed 2024-10-18 by kimtore

lidense, 2025-04-27:
Im still having the same problem of device visible only in the official app but not from spotipy. Please advise.

grainwave, 2025-08-05:
Yes confirm I'm having the same issue. Agree with @mitea1 that official spotify apps must use some other api to 'register' a device. Which is not available in the web API.

Did you ever solve this @mitea1 ?