dtcooper/raspotify #664: Music keeps on jumping (stop-play)

andrewhamili, 2024-07-10:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Music keeps on jumping. Already restarted the service but still the same.

Please see attached sceenshot from systemd status

[image]

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Jul 10 12:29:41 raspberrypi librespot[22891]: [2024-07-10T04:29:41Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink Buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EPIPE: Bro
ken pipe'
Jul 10 12:29:41 raspberrypi librespot[22891]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Jul 10 12:29:42 raspberrypi librespot[22891]: [2024-07-10T04:29:42Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink Buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EPIPE: Bro
ken pipe'
Jul 10 12:29:42 raspberrypi librespot[22891]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Jul 10 12:30:02 raspberrypi librespot[22891]: [2024-07-10T04:30:02Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink Buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EPIPE: Bro
ken pipe'
Jul 10 12:30:02 raspberrypi librespot[22891]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Jul 10 12:30:15 raspberrypi librespot[22891]: [2024-07-10T04:30:15Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink Buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EPIPE: Bro
ken pipe'
Jul 10 12:30:15 raspberrypi librespot[22891]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Jul 10 12:30:16 raspberrypi librespot[22891]: [2024-07-10T04:30:16Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink Buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EPIPE: Bro
ken pipe'
Jul 10 12:30:16 raspberrypi librespot[22891]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
```

desegura, 2024-08-05:
same problem here

SanielDous, 2024-09-02:
Same problem here. This suddenly occurred. No changes made to the system.

dtcooper, 2024-09-17:
Does latest release resolve this?

giggywithit, 2024-10-02:
> Does latest release resolve this?

nice seeing you.
I only occasionally update upgrade and will say that for whatever reason, the latest actually became unuseable.
I also run shairport alongside and am using it instead at the moment..
dmesg revealed that raspotify would open and close the audio device repeatedly until it gave up after maybe 30 sec.

kimtore, 2024-10-04:
@giggywithit will you please post those logs from raspotify and dmesg?

giggywithit, 2024-10-05:
Initially, I did an update and had the same issues, then did a full reinstall and still have the same issues...
The only thing I think it might not like is my config?

 service raspotify status
○ raspotify.service
     Loaded: masked (Reason: Unit raspotify.service is masked.)
     Active: inactive (dead)

For whatever reason, this time there was no event in trying to open the audio device... But this is the way it looked the other day.. dmesg

[ 1946.741086] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[ 1951.958341] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[ 1956.908575] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1956.908731] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1956.909423] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1956.909643] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1956.910032] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1956.910863] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[ 1961.942116] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[ 1967.082001] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1967.082162] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1967.082860] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1967.083084] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1967.083476] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1967.084308] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[ 1972.181875] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[ 1977.279098] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1977.279250] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1977.279950] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1977.280172] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1977.280561] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 1977.281391] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22
[ 1982.421638] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_mute: muting amp using GPIO22
[ 2245.570677] pcm512x 1-004c: No SCLK, using BCLK: -2
[ 2245.573188] snd-rpi-iqaudio-dac soc:sound: snd_rpi_iqaudio_gpio_unmute: un-muting amp using GPIO22

kimtore, 2024-10-06:
It would be helpful if you post your config if you think the config is the problem.

Also missing logs from raspotify itself. These dmesg messages don't look like there is an issue really.

giggywithit, 2024-10-09:
It was the config
For some reason, it didn't like the 
LIBRESPOT_AUTOPLAY=
being left empty or something...
I just commented it out and it's working so far

cat /etc/raspotify/conf 
LIBRESPOT_QUIET=
LIBRESPOT_ALSA_MIXER_DEVICE="hw:0"
LIBRESPOT_MIXER="alsa"
LIBRESPOT_ALSA_MIXER_CONTROL="Digital"
LIBRESPOT_NAME="COX-GhostAmp"
LIBRESPOT_BITRATE="320"
LIBRESPOT_INITIAL_VOLUME="20"
# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
LIBRESPOT_VOLUME_CTRL="cubic"
#LIBRESPOT_AUTOPLAY=
LIBRESPOT_DISABLE_AUDIO_CACHE=
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=
TMPDIR=/tmp

giggywithit, 2024-10-11:
Alright, now I have discovered that I have the same issues as the above users. 
My player will start and stop on it's own, usually stop, mid-song or otherwise..
My amplifier is controlled by audio being used or not, and at times it will just turn on all by itself.
I have also experienced the underrun conditions as above..
raSpotify is useless at the moment, had to uninstall

hbsagen, 2024-10-12:
Had to reboot to get it working again.

`Oct 12 20:04:39 RPi4Spotify systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Oct 12 20:06:44 RPi4Spotify librespot[13068]: [2024-10-12T18:06:44Z ERROR librespot_connect::spirc] user attributes update selected, but none received
Oct 12 20:07:14 RPi4Spotify librespot[13068]: [2024-10-12T18:07:14Z ERROR librespot_connect::spirc] user attributes mutation selected, but none received
Oct 12 20:07:44 RPi4Spotify librespot[13068]: [2024-10-12T18:07:44Z WARN  librespot_connect::spirc] Cannot flush spirc event sender when done.`

kimtore, 2024-10-17:
Please try the 0.45.0 release which contains librespot 0.5.0.

giggywithit, 2024-10-18:
> Please try the 0.45.0 release which contains librespot 0.5.0.

Seems good so far..
Thank you

kimtore, 2024-12-13:
@andrewhamili it seems other users are having luck with the proposed upgrade, please confirm whether this works for you.

enricorotundo, 2024-12-16:
fingers crossed 🤞

closed 2025-03-24 by github-actions[bot]