dtcooper/raspotify #666: Cache path option not working

fjromerom, 2024-07-27:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,

I'm seeing the following warning in the logs.

```
WARN  librespot] Without a `--cache` / `-c` path, and/or if the `--disable-audio-cache` / `-G` fl
```

I have enabled the path using `LIBRESPOT_CACHE=/tmp/` in the conf file but the warning still persists.

Also, I tried to verify raspotify is passing correctly the options I have enabled with the following command but it does not any option.

```
 raspotify ps -ef |grep libre
62638     5449     1 13 14:07 ?        00:00:53 /usr/bin/librespot
```

I have the following questions:

1. How can I verify raspotify is passing the options I have enabled if I don't see them in the process output.
2. Why the `LIBRESPOT_CACHE=/tmp/` is not working?

Thanks,
Francisco

kimtore, 2024-10-21:
Hi @fjromerom, apologies for the late reply.
Please try the latest version, Raspotify 0.45.0, and see if you can get this working.
This version includes the very last release from upstream librespot, with many fixes.

kimtore, 2024-10-21:
As for your first question, verifying input:

Every environment variable in `/etc/raspotify/conf` will be passed on to Librespot as per the `EnvironmentFile=-%E/%N/conf` line in `/lib/systemd/system/raspotify.service`.

You can review the available configuration options in the librespot wiki: https://github.com/librespot-org/librespot/wiki/Options

fjromerom, 2024-10-29:
Hi,

I've installed the latest version and I still see the same warning in the log.

```
$ dpkg -s raspotify
Package: raspotify
Status: install ok installed
Priority: optional
Section: base
Installed-Size: 15512
Maintainer: Kim Tore Jensen <kimtjen@gmail.com>
Architecture: armhf
Version: 0.45.0~librespot.v0.5.0-84d28e8
Depends: libc6 (>= 2.31), coreutils (>= 8.32), libasound2 (>= 1.2.4), alsa-utils (>= 1.2.4), libpulse0 (>= 14.2), systemd (>= 247.3), init-system-helpers (>= 1.60)
Conffiles:
 /etc/raspotify/conf 0da840e0561460d866164a39329b201f
Description: Raspotify
 A Spotify Connect systemd daemon for Debian Stable based headless systems which thinly wraps a fork of librespot.
 [Built with rustc 1.81.0]
Homepage: https://github.com/dtcooper/raspotify
Vcs-Browser: https://github.com/dtcooper/raspotify
Vcs-Git: https://github.com/dtcooper/raspotify

$ journalctl -u raspotify -r -n 3
-- Journal begins at Sun 2022-03-20 13:17:01 CET, ends at Tue 2024-10-29 17:17:20 CET. --
Oct 29 17:15:38 raspberrypi librespot[16286]: [2024-10-29T16:15:38Z WARN  librespot] Without a `--cache` / `-c` path, and/or if the `--disable-audio-cache` / `-G` f>
Oct 29 17:15:38 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
```

fjromerom, 2024-11-05:
I reported the issue [here](https://github.com/librespot-org/librespot/wiki/Options) and after commenting it out the warning went away.

closed 2024-11-05 by fjromerom