dtcooper/raspotify #668: Can't here sound from audio hat( raspberry pi 4b + Adafruit Voice Bonnet)

thenukaperea, 2024-09-03:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Im using raspberry pi 4b + Adafruit Voice Bonnet, and i followed the steps as given in the site (https://learn.adafruit.com/adafruit-voice-bonnet/blinka-setup) I got the results as mentioned in the guide lines, but I can't here anything. installed 64 bit bookworm OS
[image]

kimtore, 2024-10-15:
Too bad your audio is not working, but I don't see how Raspotify has anything to do with Adafruit Voice Bonnet?

closed 2024-10-15 by kimtore

kimtore, 2024-10-15:
Might also be related to #676