dtcooper/raspotify #669: librespot error: Unable to load encrypted file: ChannelError

sowthistle, 2024-09-03:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I've newly installed raspotify, first time I've tried it (apologies in advance if I've messed up). I've checked all packages are up to date. Able to connect but, so far, unable to play any tracks. The Spotify app it skips a number of tracks and then drops the connection. Raspotify crashes.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
seb@woody2:~ $ sudo systemctl status raspotify
○ raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; preset: enabled)
     Active: inactive (dead) since Tue 2024-09-03 11:15:42 BST; 10s ago
   Duration: 20min 18.446s
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 2057 ExecStart=/usr/bin/librespot (code=exited, status=1/FAILURE)
   Main PID: 2057 (code=exited, status=1/FAILURE)
        CPU: 886ms

Sep 03 11:15:41 woody2 librespot[2057]: [2024-09-03T10:15:41Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 11:15:41 woody2 librespot[2057]: [2024-09-03T10:15:41Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 11:15:41 woody2 librespot[2057]: [2024-09-03T10:15:41Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyI>
Sep 03 11:15:42 woody2 librespot[2057]: [2024-09-03T10:15:42Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 11:15:42 woody2 librespot[2057]: [2024-09-03T10:15:42Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 11:15:42 woody2 librespot[2057]: [2024-09-03T10:15:42Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyI>
Sep 03 11:15:42 woody2 librespot[2057]: [2024-09-03T10:15:42Z ERROR librespot_playback::player] Player::pause called from invalid state
Sep 03 11:15:42 woody2 systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Sep 03 11:15:42 woody2 systemd[1]: raspotify.service: Failed with result 'exit-code'.
Sep 03 11:15:42 woody2 systemd[1]: raspotify.service: Triggering OnFailure= dependencies.

seb@woody2:~ $ sudo cat /etc/raspotify/crash_report
-- System Info --

Linux woody2 6.6.31+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
[... 324 lines cut ...]

		Format: S16   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: S24   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: S24_3 Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: S32   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: F32   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: F64   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz

	--------------------------------------------------------------------

	Device:

		dmix:CARD=Headphones,DEV=0

	Description:

		bcm2835 Headphones, bcm2835 Headphones
		Direct sample mixing device

	Supported Format & Sample Rate Combinations:

		Format: S16   Sample Rate(s): 48kHz

	--------------------------------------------------------------------
```

Netspider, 2024-09-03:
I encountered the same error today. It worked perfectly for months (thanks for that) but today I cannot play music anymore:

```Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z WARN  librespot_core::mercury] error 400 for uri hm://autoplay-enabled/query?uri=spotify:search:***
Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: ***, audio_type: Track }>: ()
Sep 03 13:00:40 mediacenter librespot[15676]: [2024-09-03T11:00:40Z WARN  librespot_connect::spirc] No context to update from!
Sep 03 13:00:41 mediacenter librespot[15676]: [2024-09-03T11:00:41Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 13:00:41 mediacenter librespot[15676]: [2024-09-03T11:00:41Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 13:00:41 mediacenter librespot[15676]: [2024-09-03T11:00:41Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: ***, audio_type: Track }>: ()```

codyrap95, 2024-09-03:
The recommended workaround from librespot is to recompile librespot from dev as mentioned [in this comment](https://github.com/librespot-org/librespot/issues/1322#issuecomment-2325189048). 
Does anybody know how to do that from raspotify?
@JasonLG1979 Is this something you can do and push to master? Or each user has to do it manually on their machine?

d-a-v, 2024-09-04:
Hello, I propose [this fix](https://github.com/dtcooper/raspotify/compare/master...d-a-v:raspotify:raspifix)

You can compile them yourself (`make all`)

Or sillily trust me and download [one of these binaries](https://mega.nz/folder/uQdgjToa#sxXikq3gIRwiBY3LVZBRlw).

I had to make them this way (and not simply recompile upstream librespot branch dev) because my orange-pi-zero is not able to compile anything. It is now running the raspi version.

This is a temporary fix and I hope @JasonLG1979 will fix this his way.

RSKriegs, 2024-09-04:
I've attempted to mirror the [fix](https://github.com/dtcooper/raspotify/issues/669#issuecomment-2328317429) (with both Docker & Podman) but so far it doesn't work for me. I am constantly running into packaging issues at compiling part in a Dockerfile. Seems I'll stick around with Librespot for some time

d-a-v, 2024-09-04:
FWIW I ran it on ubuntu 22.04 / amd64.

aliher1911, 2024-09-04:
@RSKriegs I had two issues building:
 1 - hit disk space issues as dependency install takes at least 5GB space and I run it on low disk VM initially.
 2 - debian packaging complaining about invalid version which is caused by raspotify repo having no tags and build script injecting 'unknown'.

d-a-v, 2024-09-05:
> 2 -

You can try and add `(cd librespot; git add your-tag)` after `git clone ${REPO}` in `build.sh`

aliher1911, 2024-09-05:
That's what I did to resolve it, other than those build run ok for me on Arch/amd64. The only thing to keep in mind is that tag that you set will conflict with existing version and if you use something lower it will warn of package downgrade and if you set newer, I'm not sure how update will work once fix is published in the repo.

emc02, 2024-09-06:
For me it's working again without any changes! Maybe Spotify reverted their change or it was just part of the service down?

bvdf0251, 2024-09-06:
For me as well. Thank the heavens

codyrap95, 2024-09-06:
I wouldn't hold my breath though. Rumor has it that it's only a temporary rollback on Spotify's part. Hopefully in the meantime we can merge a working librespot version into raspotify

d-a-v, 2024-09-06:
@aliher1911 
>  I'm not sure how update will work once fix is published in the repo.

`dpkg -i newpackage.deb` will be sufficient
In case where the installer is used, which is installing an apt repository, a `dpkg -r raspotify` will remove the installed one without removing config files, then `apt install raspotify` will install the new one.

sowthistle, 2024-09-07:
> Or sillily trust me and download [one of these installers](https://mega.nz/folder/uQdgjToa#sxXikq3gIRwiBY3LVZBRlw).

I did this (please have mercy) and now listening happily. Many thanks.

closed 2024-09-17 by dtcooper

dtcooper, 2024-09-17:
Latest release 0.44.0 should address this issue. Please test!

codydubat, 2024-09-18:
@dtcooper Thanks for updating it. Sadly now there is another issue with the autplay flag, mentioned here: https://github.com/dtcooper/raspotify/issues/673

HendrikV37, 2024-10-04:
The update seems to work. Spotify finds my speaker, connects and plays music. So thanks for. I will do some further testing. W.r.t the autoplay flag, you have to use "on" or "off". e.g. LIBRESPOT_AUTOPLAY="off"

kimtore, 2024-10-18:
@HendrikV37 the autoplay default flag was fixed and the updated config is in the 0.45.0 release. Thanks for testing, did you experience any more problems?

PR #672 (2024-09-17): Compile against upstream librespot.
This patch uses upstream dev branch in favor of Jason's fork.

~The dev branch merged cleanly into that fork, and there was no diff between the two trees.~

The commit trees for the fork and upstream diverge. Many commits won't merge because of conflicts. Some code seems similar on both sides. I have no idea about the potential breakage for downstream users, but the alternative is to leave it unmaintained. Hopefully everything works fine :-)

Fixes #670
Fixes #669