dtcooper/raspotify #670: The music doesn't play, it just scrolls

krzyssiek, 2024-09-03:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I wanted to play some music but for a few days now all I've gotten is the songs scrolling to the next one and so on and so forth. Despite reinstalling the software and the entire system on the raspberry pi 4, it didn't work.

https://github.com/user-attachments/assets/f29e2d1a-015c-4866-83fd-a703d44b6acc

Just like in the movie, it changes by itself

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Sep 03 19:05:33 raspberrypi systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 53036561475767057797983400620817241747, audio_>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 126259601155372878568949598867005957397, audio>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 54447610324500953573142913273729978295, audio_>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 169521179732277629243290927173932916245, audio>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 232225039897052614237254781094114445023, audio>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 162028838171401467589615975255244950578, audio>
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:09 raspberrypi librespot[668]: [2024-09-03T17:07:09Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 140730357199516279583465159941361576804, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 208250368773906942889955416341451553453, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 5713765856195359562657769183685188797, audio_t>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 28920153809481008829636237664955493168, audio_>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 264774973186793444400250663909453871900, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 188930730237638063465794986525012297695, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 61201146394670731678103825088990558456, audio_>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 291908172769149665851632943799263535304, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 279227199881009651727523141785025869353, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 215306371885541599569871161141011996808, audio>
Sep 03 19:07:10 raspberrypi librespot[668]: [2024-09-03T17:07:10Z ERROR librespot_core::channel] channel error: 2 0
```

jonas-fischer, 2024-09-03:
same for me since a few days :(

amarelo, 2024-09-03:
It's been the same for me since last evening (GMT), while it did still work earlier in the day. It doesn't just apply to music but also podcasts, always jumping back to the start of the podcast when I attempt switching over from another Spotify device playing the same podcast already at a more advanced time position.

hbsagen, 2024-09-03:
Same problem here. Somebody said the dev version of librespot would fix it, but I don’t know how to install that so raspotify could use it.

11Maarten, 2024-09-03:
Same for me .. and I miss it a lot!

lucascajal, 2024-09-03:
See the [related issue #1322 in librespot](https://github.com/librespot-org/librespot/issues/1322)

d-a-v, 2024-09-04:
https://github.com/dtcooper/raspotify/issues/669#issuecomment-2328317429

kimtore, 2024-09-04:
Also the same issue: librespot-org/librespot#972

Would really love to see an updated version of Raspotify compiled against a working version of librespot :)

d-a-v, 2024-09-04:
An updated fix lives in https://github.com/dtcooper/raspotify/issues/669#issuecomment-2328317429

core66, 2024-09-05:
> An updated fix lives in [#669 (comment)](https://github.com/dtcooper/raspotify/issues/669#issuecomment-2328317429)

Thats not rly a fix for anyone.

core66, 2024-09-06:
**### Spotify must have realized they broke something on their end and fixed it.**

**All is working again with no changes made.**

Kybeer, 2024-09-06:
> ### **Spotify must have realized they broke something on their end and fixed it.**
> 
> _**All is working again with no changes made.**_

Same for me :)

d-a-v, 2024-09-06:
https://github.com/dtcooper/raspotify/issues/669#issuecomment-2334821383

> I wouldn't hold my breath though. Rumor has it that it's only a temporary rollback on Spotify's part. Hopefully in the meantime we can merge a working librespot version into raspotify

dtcooper, 2024-09-06:
> Also the same issue: [librespot-org/librespot#972](https://github.com/librespot-org/librespot/issues/972)
> 
> Would really love to see an updated version of Raspotify compiled against a working version of librespot :)

Our current maintainer @JasonLG1979 seems to be MIA. I'm open to others taking over!

krzyssiek, 2024-09-13:
Is there any chance it will work again in the near future?

dtcooper, 2024-09-13:
We are currently looking for a maintainer. I will update the README!

closed 2024-09-17 by dtcooper

dtcooper, 2024-09-17:
Latest release 0.44.0 should address this issue. Please test!

PR #672 (2024-09-17): Compile against upstream librespot.
This patch uses upstream dev branch in favor of Jason's fork.

~The dev branch merged cleanly into that fork, and there was no diff between the two trees.~

The commit trees for the fork and upstream diverge. Many commits won't merge because of conflicts. Some code seems similar on both sides. I have no idea about the potential breakage for downstream users, but the alternative is to leave it unmaintained. Hopefully everything works fine :-)

Fixes #670
Fixes #669