dtcooper/raspotify #671: Raspberry Pi Zero 2 W with raspotify... Simply cannot get it to work right.

dwedia, 2024-09-07:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

 I have a Raspberry Pi Zero 2 W, that I bought a few years ago. I bought it as a replacement for my Raspberry Pi Zero W, that I was using as a Spotify Connect player, when raspotify stopped supporting the old Zero W.

It worked fine for a few months, and then it just when to shit... 7 out of 10 attempts to play music through it resulted in the connect part crapping out and the song not playing. It would either "play" (the progress bar would move", but no sound was sent to my reciever, or the raspotify service would crash altogether. This would mean that I would have to try to reconnect, but often the reconnect attempts would just time out...

This happened regardless if I was using raspotify, or Librespot itself, or even volumio.

Since this had VERY low wife approval factor I ended up finding an old Chromecast Audio to use as a Spotify Connect device. This worked flawlessly (and still does...). Since the Chromecast Audio was discontinued, this solution will only work for so long....

Because of this, I tried again over the last few days, to use my Raspberry Pi Zero 2 W.

1st attempt: Raspberry Pi OS Lite, with raspotify installed. I can sometimes connect, and if I am lucky, I might get a song to play... And then it will often crash the service, and disconnect from my spotify client on my phone, thus forcing me to attempt the reconnect to spotify connect dance all over again.

2nd attempt: Installed Moode. This runs quite slowly, and it seems the Raspberry Pi Zero 2 W is underpowered for this use. The Web UI is sluggish and slow to update. I was able to play radio on it, but for spotify Connect it has the same issues as the first attempt.

3rd attempt: Installed Volumio. This would not even boot. I tried reflashing the SD card several times with both the flatpak version of Raspberry Pi Imager, and Raspberry Pi imager on Windows. I could not get it to boot. It just stops in the boot process and turns off. 

So, where do I go from here? I would ideally like to use Raspotify, with my Pi Zero 2 W, but I am at a loss as to what goes wrong...

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
journalctl: 

Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_discovery::server] POST "/" {}
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z INFO  librespot_core::session] Connecting to AP "ap2-gew4.spotify.com:4070"
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z INFO  librespot_core::session] Authenticated as "<redacteUserID>" !
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_core::session] new Session[0]
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Log(60.0)
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::player] Created <PlayerInternal> [player:0] thread
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_connect::spirc] new Spirc[0]
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_connect::spirc] canonical_username: <redacteUserID>
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot::component] new MercuryManager
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::mixer::mappings] Input volume 58958 mapped to: 49.99%
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16, sample rate: 44100
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::resampler] Sample Rate: 44.1kHz
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::resampler] Interpolation Type: Bypass
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::resampler] No <ResampleWorker> threads required
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::normaliser] Normalisation Type: Auto
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::normaliser] Normalisation Pregain: 0.0 dB
Sep 07 22:00:07 Barry librespot[9164]: [2024-09-07T20:00:07Z DEBUG librespot_playback::normaliser] Normalisation Threshold: -2.0 dBFS
[... 342 lines cut ...]
		Format: S24_3 Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: S32   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: F32   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz
		Format: F64   Sample Rate(s): 44.1kHz, 48kHz, 88.2kHz, 96kHz

	--------------------------------------------------------------------

	Device:

		hdmi:CARD=vc4hdmi,DEV=0

	Description:

		vc4-hdmi, MAI PCM i2s-hifi-0
		HDMI Audio Output

	Supported Format & Sample Rate Combinations:

		Format: S16   Sample Rate(s): 44.1kHz, 48kHz
		Format: S24   Sample Rate(s): 44.1kHz, 48kHz
		Format: S24_3 Sample Rate(s): 44.1kHz, 48kHz
		Format: S32   Sample Rate(s): 44.1kHz, 48kHz

	--------------------------------------------------------------------
```

kimtore, 2024-09-17:
Hi @dwedia 

Hopefully there will be a new release out soon with updates.
I think that might have the highest probability of success...

In the mean time, and this is a long shot, perhaps try to add this to your `/etc/hosts` file:
```
0.0.0.0 apresolve.spotify.com
```

dtcooper, 2024-09-17:
According to @kimtore, latest release 0.44.0 should address this issue. Please test!

dwedia, 2024-09-22:
> According to @kimtore, latest release 0.44.0 should address this issue. Please test!

I am updating as I am typing this. I will put it to the test.
Thanks for the hard work =)

dwedia, 2024-09-23:
> Hi @dwedia
> 
> Hopefully there will be a new release out soon with updates. I think that might have the highest probability of success...
> 
> In the mean time, and this is a long shot, perhaps try to add this to your `/etc/hosts` file:
> 
> ```
> 0.0.0.0 apresolve.spotify.com
> ```

I added this, after I posted the issue first (it was recomended to me on the raspberry pi subreddit). 
I can see it failing to connect in the systemd journal (as in the workaround is working ;) ). 

`Sep 23 17:55:02 Barry librespot[531]: [2024-09-23T15:55:02Z WARN  librespot_core::apresolve] Failed to resolve all access points, using fallbacks
Sep 23 17:55:02 Barry librespot[531]: [2024-09-23T15:55:02Z WARN  librespot_core::apresolve] Resolve access points error: Service unavailable { client error (Connect) }
Sep 23 17:55:02 Barry librespot[531]: [2024-09-23T15:55:02Z WARN  librespot_core::apresolve] Failed to resolve all access points, using fallbacks
Sep 23 17:55:02 Barry librespot[531]: [2024-09-23T15:55:02Z WARN  librespot_core::apresolve] Resolve access points error: Service unavailable { client error (Connect) }
Sep 23 17:55:03 Barry librespot[531]: [2024-09-23T15:55:03Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
`

> > According to @kimtore, latest release 0.44.0 should address this issue. Please test!
> 
> I am updating as I am typing this. I will put it to the test. Thanks for the hard work =)

I have tested quite a bit, both yesterday and today.
What I have tested is:
 - connect and disconnect from phone in spotify.
 - Play, stop, change songs in spotify.
 - connect and disconnect from spotify on my computer.
 - connect from a different spotify account (on another phone), while spotify plays from my account, and then "taking the stream back to my account".

Everything seems to work now :)

Again, thank you for the work you do on this

kimtore, 2024-09-25:
Very nice. Thanks for the feedback. :)

closed 2024-09-25 by kimtore