dtcooper/raspotify #673: Invalid --autoplay / -A in librespot

TupaNegreiros, 2024-09-18:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Raspotify worked for years and stopped today.

service raspotify status shows:
set 18 09:38:08 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
set 18 09:38:08 raspberrypi librespot[7213]: [2024-09-18T12:38:08Z ERROR librespot] Invalid `--autoplay` / `-A`: ""
set 18 09:38:08 raspberrypi librespot[7213]: Valid `--autoplay` / `-A` values: on, off
set 18 09:38:08 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
set 18 09:38:08 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
set 18 09:38:08 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
-- System Info --

Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

-- Logs --

set 18 09:37:10 raspberrypi librespot[5420]: [2024-09-18T12:37:05Z ERROR librespot] Invalid `--autoplay` / `-A`: ""
set 18 09:37:10 raspberrypi librespot[5420]: Valid `--autoplay` / `-A` values: on, off
set 18 09:37:05 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
set 18 09:37:05 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
set 18 09:37:05 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
set 18 09:37:21 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
set 18 09:37:21 raspberrypi librespot[5784]: [2024-09-18T12:37:21Z ERROR librespot] Invalid `--autoplay` / `-A`: ""
set 18 09:37:21 raspberrypi librespot[5784]: Valid `--autoplay` / `-A` values: on, off
set 18 09:37:21 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
set 18 09:37:21 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
set 18 09:37:21 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
[... 198 lines cut ...]

                vc4-hdmi, MAI PCM i2s-hifi-0
                Default Audio Device

        Supported Format(s):

                S16 S24 S24_3 S32 F32 F64

        ------------------------------------------------------

        Device:

                hdmi:CARD=vc4hdmi,DEV=0

        Description:

                vc4-hdmi, MAI PCM i2s-hifi-0
                HDMI Audio Output

        Supported Format(s):

                S16 S24 S24_3 S32

        ------------------------------------------------------
```

codydubat, 2024-09-18:
Having the same issue on RPi 3b+ after the 0.44 update.
`Sep 18 16:06:01 raspberrypi systemd[1]: raspotify-crash-report-generator.service: Consumed 1.562s CPU time.
Sep 18 16:06:01 raspberrypi librespot[9095]: [2024-09-18T13:06:01Z ERROR librespot] Invalid `--autoplay` / `-A`: ""
Sep 18 16:06:01 raspberrypi librespot[9095]: Valid `--autoplay` / `-A` values: on, off
Sep 18 16:06:01 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 16:06:01 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Sep 18 16:06:01 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Sep 18 16:06:01 raspberrypi systemd[1]: Starting Raspotify Crash Report Generator...`

bvdf0251, 2024-09-18:
I can confirm I've got the same issue.

Also huge thanks for the effort non the less

codydubat, 2024-09-18:
If you want a quick fix for your local environment open `/etc/raspotify/conf` and just set LIBRESPORT_AUTOPLAY to `on` or `off`:
`LIBRESPOT_AUTOPLAY=on`

TupaNegreiros, 2024-09-18:
> LIBRESPORT_AUTOPLAY

I confirm adding `LIBRESPOT_AUTOPLAY="on"` at the end of `/etc/raspotify/conf` fixed.

codydubat, 2024-09-18:
@bvdf0251 For me commenting it also works, you just need to run `sudo systemctl restart raspotify` in order for the config to be re-applied

bvdf0251, 2024-09-18:
With this addition it's showing me an error 'failed to register ipv6 receiver' and I can't connect from my phone.

XuluWarrior, 2024-09-18:
I'm assuming that this is an issue with 0.44.0 after #672
As everything was working for me before upgrading from 0.43.x (can't remember the exact version)

If so, can we include this a requirement in the release notes.

bvdf0251, 2024-09-18:
> With this addition it's showing me an error 'failed to register ipv6 receiver' and I can't connect from my phone.

Fixed by enabling ipv6 

Still not working though. It's starting but unable to connect from phone.

Downgrade to latest 0.43 still working.

PR #674 (2024-09-20): Update configuration template for upstream 0.5.0
* Removes LIBRESPOT_SAMPLE_RATE
* Removes LIBRESPOT_INTERPOLATION_QUALITY
* Sets boolean values in config to 'on' or 'off'

Fixes #673
closed 2024-09-20 by dtcooper