dtcooper/raspotify #675: Enable support for OAuth

1ukastesar, 2024-09-19:
### Raspotify is just a package

- [X] I know Raspotify is just a package

### Due Diligence

- [X] I have done my due diligence

### What can we do to make Raspotify more awesome?

As of https://github.com/librespot-org/librespot/issues/1308 and subsequent PR https://github.com/librespot-org/librespot/pull/1309, Spotify has apparently disabled password auth altogether and forces OAuth for everyone. Please add an option to authenticate with OAuth through `librespot`. Upstream `librespot` already supports OAuth with `-j` option, somehow I can't figure out which cache dir to use to get it working after successful auth.

TagorePde, 2024-09-19:
Same issue i get :

`Password authentication no longer supported, use OAuth`

any way to solve this?
Regards.

krzyssiek, 2024-09-23:
i also have this problem.

sanjibm, 2024-09-24:
Same problem, got the auth token via librespot, but not sure where to put it so that Raspotify can see it

kimtore, 2024-09-25:
I suspect this can be solved by not using a dynamic user through systemd. This way, you will be able to write to raspotify's directory. I won't be able to test this until next week, however. Perhaps you could try it out.

Change `/lib/systemd/system/raspotify.service` from
```
[Service]
DynamicUser=yes
```
to
```
[Service]
DynamicUser=yes
```
Then run `systemctl daemon-reload && systemctl restart raspotify`.

sanjibm, 2024-09-25:
Thanks @kimtore, that worked perfectly!

For anyone landing here, below is a summary of the steps I followed to get this working on a Raspberry Pi 4 running Bullseye:

1. Stopped the raspotify service with `systemctl stop raspotify` after noticing errors on the librespot logs: _Connection failed: Login failed with reason: Bad credentials_

2. Cloned the tip of the librespot repository to get the fix (no new release available at the time of writing, v0.4.2 was last release) and built it locally on the Raspberry Pi following instructions here:
https://github.com/librespot-org/librespot/blob/master/COMPILING.md

Might need to install Rust if you didn't have it already. Make sure you select the right backend, in my case had to specify alsa using: `cargo build --no-default-features --features "alsa-backend"`

4. Followed the advice in this comment to authenticate using OAuth: https://github.com/librespot-org/librespot/pull/1309#issuecomment-2348563401

After running the `librespot --cache <YOUR_CONFIGURATION_PATH> -j` command on the Pi, I noted the listener port number

5. Then copied the full output URL on to a laptop browser, which had port-forwarding setup via ssh to point back to the Raspberry Pi IP address and port. 

6. That then authenticated to Spotify on the laptop browser (needed to enter Spotify username/password), which sent the authentication response back to the Raspberry Pi (via port-forwarded), writing the authentication token to the path specified when running librespot with the -j option

7. Changed `/lib/systemd/system/raspotify.service` as advised above to specify

> [Service]
> DynamicUser=no

Then run `systemctl daemon-reload && systemctl restart raspotify`.

After this, noticed below new lines in the librespot logs:
_Found pre-existing private CacheDirectory= directory /var/cache/private/raspotify, migrating to /var/cache/raspotify.
Apparently, service previously had DynamicUser= turned on, and has now turned it off._

Was also still seeing the Bad Credentials failure message too at this point, as it was still using username/password

8. The authentication token was still in the folder specified and the token file is called _credentials.json_. I copied that now to the cache folder mentioned in the logs: `cp credentials.json /var/cache/raspotify/`

9. Updated the raspotify config file `sudo vi /etc/raspotify/conf` to comment out below items to enable 0Auth instead of username/password

> #LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
> #LIBRESPOT_DISABLE_DISCOVERY=
> #LIBRESPOT_USERNAME="mysecretusername"
> #LIBRESPOT_PASSWORD="mysecretpassword"

10. Restarted raspotify with `systemctl restart raspotify` and everything worked, clean librespot logs no more errors

kimtore, 2024-09-26:
Great, thanks for detailing the steps!

Here are my thoughts.

* Manual building steps should not be needed for Raspotify 0.44.1, because it already has the oauth feature merged in.
* Everyone agrees the OAuth flow user experience is suboptimal, but the conversation on librespot's side is not resolved yet.
* In the meantime, I suggest changes to the systemd service, configuration file and documentation.

ACalleM, 2024-11-10:
Just wanted to comment that you dont need port forwarding for the OAuth flow. You can copy the callback URL returned by spotify (with a 127.0.0.1 address) and use curl to invoke it from the raspberrypi where you executed "librespot --cache <YOUR_CONFIGURATION_PATH> -j"

closed 2025-02-26 by github-actions[bot]

snadeemshaikh, 2025-03-19:
> Thanks [@kimtore](https://github.com/kimtore), that worked perfectly!
> 
> For anyone landing here, below is a summary of the steps I followed to get this working on a Raspberry Pi 4 running Bullseye with Librespot v0.4.2. Note that although I had to compile librespot from source to run it and get the authentication token, I did not need to actually install it again overwriting the release version.
> 
> 1. Stopped the raspotify service with `systemctl stop raspotify` after noticing errors on the librespot logs: _Connection failed: Login failed with reason: Bad credentials_
> 2. Cloned the tip of the librespot repository to get the fix (no new release available at the time of writing, v0.4.2 was last release) and built it locally on the Raspberry Pi following instructions here:
>    https://github.com/librespot-org/librespot/blob/master/COMPILING.md
> 
> Might need to install Rust if you didn't have it already. Make sure you select the right backend, in my case had to specify alsa using: `cargo build --no-default-features --features "alsa-backend"`
> 
> 4. Followed the advice in this comment to authenticate using OAuth: [Credentials with access token (oauth) librespot-org/librespot#1309 (comment)](https://github.com/librespot-org/librespot/pull/1309#issuecomment-2348563401)
> 
> After running the `librespot --cache <YOUR_CONFIGURATION_PATH> -j` command on the Pi, I noted the listener port number
> 
> 5. Then copied the full output URL on to a laptop browser, which had port-forwarding setup via ssh to point back to the Raspberry Pi IP address and port.
> 6. That then authenticated to Spotify on the laptop browser (needed to enter Spotify username/password), which sent the authentication response back to the Raspberry Pi (via port-forwarding), writing the authentication token to the path specified when running librespot with the -j option
> 7. Changed `/lib/systemd/system/raspotify.service` as advised above to specify
> 
> ```
> [Service]
> DynamicUser=no
> ```
> 
> Then run `systemctl daemon-reload && systemctl restart raspotify`.
> 
> After this, noticed below new lines in the librespot logs: _Found pre-existing private CacheDirectory= directory /var/cache/private/raspotify, migrating to /var/cache/raspotify. Apparently, service previously had DynamicUser= turned on, and has now turned it off._
> 
> Was also still seeing the Bad Credentials failure message too at this point, as it was still using username/password
> 
> 8. The authentication token was still in the folder specified and the token file is called _credentials.json_. I copied that now to the cache folder mentioned in the logs: `cp credentials.json /var/cache/raspotify/`
> 9. Updated the raspotify config file `sudo vi /etc/raspotify/conf` to comment out below items to enable 0Auth instead of username/password
> 
> ```
> #LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
> #LIBRESPOT_DISABLE_DISCOVERY=
> #LIBRESPOT_USERNAME="mysecretusername"
> #LIBRESPOT_PASSWORD="mysecretpassword"
> ```
> 
> 10. Restarted raspotify with `systemctl restart raspotify` and everything worked, clean librespot logs no more errors

Hi Sanjibm, my team needs help on this and is ready to pay for your time as well. Can you reach out to nadeem@naptick.com or share your contact to connect?