dtcooper/raspotify #676: Compatible alsa device(s): empty

y-lobau, 2024-09-19:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

**Setup:**
- RPI 5 on a bookworm
- librespot 0.5.0-dev
- raspotify 0.44.0

**Problem:**
Raspotify is running but list of supported ALSA devices is empty. What may cause this?
 
```shell
sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-09-19 21:44:09 BST; 8s ago
```  
```shell
rpi@rpi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: 1f000a0000.i2s-wm8960-hifi wm8960-hifi-0 [1f000a0000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
``` 

```shell
librespot -d ?
[2024-09-19T20:45:37Z INFO  librespot] librespot 0.5.0-dev fb5c0ef (Built on 2024-09-17, Build ID: jyRJ1a0M, Profile: release)

        Compatible alsa device(s):

        ------------------------------------------------------
``` 

Then the Raspotify is not recognized by Spotify connect, and I assume this may be because there's no available device for librespot.

kimtore, 2024-09-20:
Not exactly sure what this could be, your ALSA setup looks fine.
Perhaps the ALSA device is claimed by some other software running simultaneously?

kletze, 2024-10-03:
@y-lobau  o you use the PI 5 without anything connected? I mean the bare board as you get it from the supplier.

YokaiRick, 2024-10-11:
Just as a heads-up, i had the same problem, but executing it with sudo the device lists outputs my devices.
so it could be a permission problem

kimtore, 2024-10-18:
@y-lobau please try again with 0.45.0.

y-lobau, 2024-10-20:
Unfortunately, the same issue with the latest raspotify/librespot, checked on the rpi zero 2 w: 

        $ sudo librespot -d ?
        [2024-10-20T15:15:15Z INFO  librespot] librespot 0.5.0 84d28e8 (Built on 2024-10-17, Build ID: CmvEuO4M, Profile: release)

        Compatible alsa device(s):

        ------------------------------------------------------

        ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave

y-lobau, 2024-10-21:
I agree, it looks like my infra issue. Indeed, the rpi also runs the active echo cancellation built into the pipewire. Also, my app is actively listening for the microphone. Maybe there's a clash somewhere in this combination.
My hope is that anyone might face with similar issue.

kimtore, 2024-10-21:
Ah, you're running Pipewire on top? Then Pipewire is most certainly holding your audio device.

Might I suggest installing `pipewire-pulse` and try to run your audio stuff over Pulseaudio instead? This way you get all the software mixing built-in from Pipewire.

y-lobau, 2024-11-11:
This issue is similar to other open issues connected with the AlsaSink. When I run the librespot manually, It is discoverable and works fine. However, the Raspotify service runs into some permissions issues when attempting to play smth with it:

Nov 11 15:31:52 myadam systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Nov 11 15:33:02 myadam librespot[4862]: Failed to create secure directory (/root/.config/pulse): Permission denied
Nov 11 15:33:02 myadam librespot[4862]: [2024-11-11T13:33:02Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device hw:1,0 May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Device or resource busy (16)'

closed 2024-11-11 by y-lobau

kimtore, 2024-11-20:
Thanks for replying. I see that there is a permission issue related to PulseAudio issue, which is be tracked in #686.
For Alsa I'm really not sure what is causing this, perhaps the service is too constrained and we might need to introduce a dedicated system user.