dtcooper/raspotify #679: Connection timed out (os error 110)

Napoletana, 2024-10-12:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I'm getting an error on raspotify after a long overdue update cycle: 

0:00: Started Raspotify (Spotify Connect Client).
2:20: librespot[5021]: [2024-10-12T18:13:48Z ERROR librespot] could not initialize spirc: Deadline expired before operation could complete { Connection timed out (os error 110) }

after which:
systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE

I've tried manually running librespot, but that didn't work either with the same error. 
I've tried purging and reinstalling raspotify
I've tried adjusting the DNS in pihole to connect to a different spotify AP
Manually added credentials to the configuration didn't work either. 

The only posts with error 110 that I can see were 'magically' fixed...

kimtore, 2024-10-15:
> I've tried manually running librespot, but that didn't work either with the same error.

Then it is probably not a Raspotify error, because Raspotify is only a wrapper around librespot. So probably best to ask librespot devs, because I really don't know about your error.

Also, not sure what you're referring to about magically fixed error 110?

Try not using your PiHole and see if that helps.

kimtore, 2024-10-17:
Perhaps you'd like to try the 0.45.0 release which contains librespot 0.5.0 with many bug fixes.

Napoletana, 2024-10-19:
I can confirm, the update (0.44.1 --> 0.45.0) fixed the issue. Thanks a lot!

closed 2024-10-20 by kimtore