dtcooper/raspotify #681: Please roll out a new release including librespot 0.5.0

philBrown, 2024-10-16:
### Raspotify is just a package

- [X] I know Raspotify is just a package

### Due Diligence

- [X] I have done my due diligence

### What can we do to make Raspotify more awesome?

I've been keeping an eye on https://github.com/librespot-org/librespot/issues/1340 and now that [v0.5.0](https://github.com/librespot-org/librespot/releases/tag/v0.5.0) has been released was going to raise a PR here to bump the version; looks like I was [beaten to the punch](https://github.com/dtcooper/raspotify/commit/e43bc717a9b40b1e8b143c2b534127b689dbaf02) by @kimtore.

Any chance you could roll out a new release of raspotify? The stability fixes in librespot would be most appreciated

kimtore, 2024-10-17:
I would release a new version but I need some help from @dtcooper to do that.

kimtore, 2024-10-17:
Also, 0.44.1 includes a version of librespot that is quite close to the released 0.5.0, have you tried that one?

philBrown, 2024-10-17:
I have indeed. I've been getting lots of unrecoverable session drops and the latest release addresses that specifically (see the librespot issue I linked).

kimtore, 2024-10-17:
Aha, yes, I see.

I can't make a release, but the workaround for you could be to install `docker` on your computer, check out the repository, and run `make` to get a debian package.

kimtore, 2024-10-17:
0.45.0 has been released with librespot 0.5.0.

closed 2024-10-17 by kimtore

philBrown, 2024-10-17:
Thanks so much! ❤️