dtcooper/raspotify #685: Raspotify has stopped playing tracks

olivier-boesch, 2024-10-27:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

raspotify was running on two rpi 3b+ with the pi os lite 64bits
only raspotify is installed on top of the os.

below is the status of the service when i try to play a song on the device

the config file was not modified (in a second test) - > same result

Any idea ?

Thanks.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-10-27 17:33:38 GMT; 33min ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 1925 (librespot)
      Tasks: 6 (limit: 762)
        CPU: 1.051s
     CGroup: /system.slice/raspotify.service
             └─1925 /usr/bin/librespot

Oct 27 17:33:38 mixscene systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z ERROR librespot_connect::spirc] ContextError: Error { kind: InvalidArgument, error: StatusCode(400) }
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z WARN  librespot_core::cdn_url] Cannot parse CDN URL expiry timestamp 'verify=1730136868-%2BJ2xRHvx7KEqBocXdlTZmO5%2Fn2v%2FhPCTiVlkg2Cw%2BMI%3D' from 'https://audio-cf-del-574.spotifycdn.com/audio/668e55eeae8b424ae6595cdae4f48ee9c3c4e02c?verify=1730136868-%2BJ2xRHvx7KEqBocXdlTZmO5%2Fn2v%2FhPCTiVlkg2Cw%2BMI%3D'
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z ERROR librespot_playback::player] Unable to load encrypted file: Error { kind: Unavailable, error: hyper_util::client::legacy::Error(Connect, Custom { kind: Other, error: Custom { kind: Other, error: Custom { kind: InvalidData, error: AlertReceived(HandshakeFailure) } } }) }
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId("spotify:track:47oS7xB31QQUyPCgHpM3VZ")>: ()
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z WARN  librespot_core::cdn_url] Cannot parse CDN URL expiry timestamp 'verify=1730136868-ruquHGqShftoa4A8rgAfPAyVWWcNONyejpcYQM4C2fk%3D' from 'https://audio-cf-del-574.spotifycdn.com/audio/2e5b8fa346611771e426acbc84158a2185554345?verify=1730136868-ruquHGqShftoa4A8rgAfPAyVWWcNONyejpcYQM4C2fk%3D'
Oct 27 17:34:28 mixscene librespot[1925]: [2024-10-27T17:34:28Z ERROR librespot_playback::player] Unable to load encrypted file: Error { kind: Unavailable, error: hyper_util::client::legacy::Error(Connect, Custom { kind: Other, error: Custom { kind: Other, error: Custom { kind: InvalidData, error: AlertReceived(HandshakeFailure) } } }) }
```

DanielBie, 2024-10-27:
Hello,
I've been having the same issue for a few hours now. From what I've gathered it's connected to some issue in librespot. At least here they talk about the same  problem:
[https://github.com/librespot-org/librespot/issues/1322#issue-2501292753](url)

I've not been able to get it working again myself yet, so any help would be appreciated.
Thank you :)

richardhapb, 2024-10-27:
Hello, I have the same error, the warning 

``` bash
Cannot parse CDN URL expiry timestamp 'verify=1730136868-%2BJ2xRHvx7KEqBocXdlTZmO5%2Fn2v%2FhPCTiVlkg2Cw%2BMI%3D' from 'https://audio-cf-del-574.spotifycdn.com/audio/668e55eeae8b424ae6595cdae4f48ee9c3c4e02c?verify=1730136868-%2BJ2xRHvx7KEqBocXdlTZmO5%2Fn2v%2FhPCTiVlkg2Cw%2BMI%3D'
```

has been produced because Librespot is looking for the expiry timestamp in the url with "__token__" or exp string but Spotify returns verify=****

First need clone the repository:

```bash
git clone https://github.com/librespot-org/librespot.git
```
I changed the conditional in the function for url verification, in file `core/src/cdn_url.rs` from this (line 119), note that I only changed the conditional:

```rust
impl TryFrom<CdnUrlMessage> for MaybeExpiringUrls {
    type Error = crate::Error;
    fn try_from(msg: CdnUrlMessage) -> Result<Self, Self::Error> {
        if !matches!(
            msg.result.enum_value_or_default(),
            StorageResolveResponse_Result::CDN
        ) {
            return Err(CdnUrlError::Storage.into());
        }

        let is_expiring = !msg.fileid.is_empty();

        let result = msg
            .cdnurl
            .iter()
            .map(|cdn_url| {
                let url = Url::parse(cdn_url)?;
                let mut expiry: Option<Date> = None;

                if is_expiring {
                    let mut expiry_str: Option<String> = None;
                    if let Some(token) = url
                        .query_pairs()
                        .into_iter()
                        .find(|(key, _value)| key == "verify")
                    {
                        // El valor del parámetro 'verify' tiene el formato 'timestamp-signature'
                        if let Some((expiry_str_candidate, _)) = token.1.split_once('-') {
                            expiry_str = Some(expiry_str_candidate.to_string());
                        }
                    } else if let Some(token) = url
                        .query_pairs()
                        .into_iter()
                        .find(|(key, _value)| key == "__token__")
                    {
                        //"https://audio-ak-spotify-com.akamaized.net/audio/4712bc9e47f7feb4ee3450ef2bb545e1d83c3d54?__token__=exp=1688165560~hmac=4e661527574fab5793adb99cf04e1c2ce12294c71fe1d39ffbfabdcfe8ce3b41",
                        if let Some(mut start) = token.1.find("exp=") {
                            start += 4;
                            if token.1.len() >= start {
                                let slice = &token.1[start..];
                                if let Some(end) = slice.find('~') {
                                    // this is the only valid invariant for akamaized.net
                                    expiry_str = Some(String::from(&slice[..end]));
                                } else {
                                    expiry_str = Some(String::from(slice));
                                }
                            }
                        }
                    } else if let Some(token) = url
                        .query_pairs()
                        .into_iter()
                        .find(|(key, _value)| key == "Expires")
                    {
                        //"https://audio-gm-off.spotifycdn.com/audio/4712bc9e47f7feb4ee3450ef2bb545e1d83c3d54?Expires=1688165560~FullPath~hmac=IIZA28qptl8cuGLq15-SjHKHtLoxzpy_6r_JpAU4MfM=",
                        if let Some(end) = token.1.find('~') {
                            // this is the only valid invariant for spotifycdn.com
                            let slice = &token.1[..end];
                            expiry_str = Some(String::from(&slice[..end]));
                        }
                    } else if let Some(query) = url.query() {
                        //"https://audio4-fa.scdn.co/audio/4712bc9e47f7feb4ee3450ef2bb545e1d83c3d54?1688165560_0GKSyXjLaTW1BksFOyI4J7Tf9tZDbBUNNPu9Mt4mhH4=",
                        let mut items = query.split('_');
                        if let Some(first) = items.next() {
                            // this is the only valid invariant for scdn.co
                            expiry_str = Some(String::from(first));
                        }
                    }

                    if let Some(exp_str) = expiry_str {
                        if let Ok(expiry_parsed) = exp_str.parse::<i64>() {
                            if let Ok(expiry_at) = Date::from_timestamp_ms(expiry_parsed * 1_000) {
                                let with_margin = expiry_at.saturating_sub(CDN_URL_EXPIRY_MARGIN);
                                expiry = Some(Date::from(with_margin));
                            }
                        } else {
                            warn!("Cannot parse CDN URL expiry timestamp '{exp_str}' from '{cdn_url}'");
                        }
                    } else {
                        warn!("Unknown CDN URL format: {cdn_url}");
                    }
                }
                Ok(MaybeExpiringUrl(cdn_url.to_owned(), expiry))
            })
            .collect::<Result<Vec<MaybeExpiringUrl>, Error>>()?;

        Ok(Self(result))
    }
}
```

And recompile (you need stay in the Librespot project folder):

```
cargo build --release
sudo cp target/release/librespot /usr/local/bin
```

This solve the warning but the `Unable to load encrypted file` error is not solved. I have the same issue, the issue is only en some tracks, another works ok.

olivier-boesch, 2024-10-28:
I simply downgraded to v.0.43.56. All worls well. 
I think I'll go on like this until the bug is resolved.

You're right, librespot seems the problem.

Thanks for your answer.

soldevi, 2024-10-29:
Upgraded some Libdrm packages today and raspotify's playing again over here. Raspotify's on version 0.45.

DanielBie, 2024-10-29:
For me it just randomly started working again yesterday, didn't change anything. Hoping it'll just continue to work now.

gklingler, 2024-10-29:
Can confirm it started working again after running an "apt update && apt upgrade" on my raspberry pi, what contained some drm and crypto packages:

```
...
The following packages will be upgraded:
  base-files bluez firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek
  initramfs-tools initramfs-tools-core libaom3 libbluetooth3 libc-bin libc-dev-bin libc-devtools libc-l10n libc6
  libc6-dbg libc6-dev libcamera-ipa libcamera0.3 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1
  libdrm2 libgssapi-krb5-2 libheif1 libk5crypto3 libkrb5-3 libkrb5support0 libpam-systemd libpisp-common libpisp1
  libssl3 libsystemd-shared libsystemd0 libudev1 libwayland-client0 libwayland-cursor0 libwayland-egl1
  libwayland-server0 linux-headers-rpi-2712 linux-headers-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v8
  linux-libc-dev locales openssl python3-gpiozero raspberrypi-sys-mods raspi-config raspi-firmware raspi-utils
  rpi-eeprom rpicam-apps-lite systemd systemd-sysv systemd-timesyncd udev userconf-pi wpasupplicant
```

olivier-boesch, 2024-11-05:
Yes, the latest version is now working...

Thanks for your help

closed 2024-11-05 by olivier-boesch

Emal011, 2025-08-02:
I'm having the same issue as described. Everything was working fine, until a couple of days, after an update in my pi. Since then I'm not able to use spotfy. Airplay is working, so the sound card is ok.
I searched all over and tried everything. I remove (purge) spotify and installed again, nothing.

`● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2025-08-02 16:38:00 CEST; 23s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 1498 (librespot)
      Tasks: 8 (limit: 2137)
        CPU: 223ms
     CGroup: /system.slice/raspotify.service
             └─1498 /usr/bin/librespot

Aug 02 16:38:00 HiFi-Wi systemd[1]: Started Raspotify (Spotify Connect Client).
Aug 02 16:38:18 HiFi-Wi librespot[1498]: [2025-08-02T14:38:18Z WARN  librespot_core::cdn_url] Cannot parse CDN URL expiry timestamp 'verify=1754231898-b0P%2BL0CCuBu7GuaLoIG87KdosoWAowmz06Ka3HBbWRE%3D' from 'https://audio-cf.spotifycdn.com/audio/3b9eb624dba30220096441283872dc92bfee9571?verify=1754231898-b0P>
Aug 02 16:38:18 HiFi-Wi librespot[1498]: [2025-08-02T14:38:18Z ERROR librespot_playback::player] Unable to load encrypted file: Error { kind: Unavailable, error: hyper_util::client::legacy::Error(Connect, Custom { kind: Other, error: ConnectError("dns error", Custom { kind: Uncategorized, error: "failed to>
Aug 02 16:38:18 HiFi-Wi librespot[1498]: [2025-08-02T14:38:18Z ERROR librespot_playback::player] Skipping to next track, unable to load track <SpotifyId("spotify:track:0mf2bwgS0JlL425FqLWkp6")>: ()
Aug 02 16:38:19 HiFi-Wi librespot[1498]: [2025-08-02T14:38:19Z WARN  librespot_core::cdn_url] Cannot parse CDN URL expiry timestamp 'verify=1754231899-FsH7DEaIYH19ovb51sdEcKjrVzLTQ8F4pd9IvHG7tU4%3D' from 'https://audio-cf.spotifycdn.com/audio/b2d7430597165b1d9b2dff7cefb38afc259bd6ed?verify=1754231899-FsH7D>
Aug 02 16:38:19 HiFi-Wi librespot[1498]: [2025-08-02T14:38:19Z ERROR librespot_playback::player] Unable to load encrypted file: Error { kind: Unavailable, error: hyper_util::client::legacy::Error(Connect, Custom { kind: Other, error: ConnectError("dns error", Custom { kind: Uncategorized, error: "failed to>`