dtcooper/raspotify #686: Raspotify will not run.

twbonnot, 2024-11-01:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I have uninstalled, reinstalled, tried multiple different methods and although it shows up, sometimes connects, it will not play. I click play the timeline doesn't move and it doesn't play anything. It immediately changes back to the pause button. Tried different devices- phone, web, computer. But Nothing works... Please guide me it's my first time using a Ras Pi.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Oct 31 21:44:25 raspberrypi systemd[1]: Started raspotify.service - Raspotify (>
Oct 31 21:45:53 raspberrypi librespot[729]: [2024-11-01T01:45:53Z ERROR libresp>
Oct 31 21:45:53 raspberrypi librespot[729]: [2024-11-01T01:45:53Z WARN  libresp>
Oct 31 21:45:55 raspberrypi librespot[729]: Failed to create secure directory (>
Oct 31 21:45:55 raspberrypi librespot[729]: [2024-11-01T01:45:55Z ERROR libresp>
Oct 31 21:45:55 raspberrypi librespot[729]: [2024-11-01T01:45:55Z ERROR libresp>
Oct 31 21:45:55 raspberrypi librespot[729]: [2024-11-01T01:45:55Z ERROR libresp>
Oct 31 21:45:55 raspberrypi systemd[1]: raspotify.service: Main process exited,>
Oct 31 21:45:55 raspberrypi systemd[1]: raspotify.service: Failed with result '>
Oct 31 21:45:55 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure>
Oct 31 21:46:06 raspberrypi systemd[1]: Started raspotify.service - Raspotify (>
Oct 31 21:46:44 raspberrypi librespot[2640]: [2024-11-01T01:46:44Z WARN  libres>
Oct 31 21:46:46 raspberrypi librespot[2640]: Failed to create secure directory >
Oct 31 21:46:46 raspberrypi librespot[2640]: [2024-11-01T01:46:46Z ERROR libres>
Oct 31 21:46:47 raspberrypi librespot[2640]: [2024-11-01T01:46:47Z ERROR libres>
Oct 31 21:47:42 raspberrypi librespot[2640]: [2024-11-01T01:47:42Z ERROR libres>
Oct 31 21:47:44 raspberrypi librespot[2640]: [2024-11-01T01:47:44Z ERROR libres>
Oct 31 21:48:48 raspberrypi librespot[2640]: [2024-11-01T01:48:48Z ERROR libres>
Oct 31 21:48:51 raspberrypi librespot[2640]: [2024-11-01T01:48:51Z ERROR libres>
Oct 31 21:48:52 raspberrypi librespot[2640]: [2024-11-01T01:48:52Z ERROR libres>
Oct 31 21:48:54 raspberrypi librespot[2640]: [2024-11-01T01:48:54Z ERROR libres>
Oct 31 21:48:55 raspberrypi librespot[2640]: [2024-11-01T01:48:55Z ERROR libres>
Oct 31 21:48:59 raspberrypi librespot[2640]: [2024-11-01T01:48:59Z ERROR libres>
```

kimtore, 2024-11-04:
Hi, maybe I can help but can you post a log which is readable? Right now it's not really useful.

nilswiersma, 2024-11-07:
I think I am having the same issue, here is a better log:

```
Nov 07 15:28:48 piathome librespot[1918]: [2024-11-07T15:28:48Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
Nov 07 15:28:48 piathome librespot[1918]: Failed to create secure directory (/root/.config/pulse): Permission denied
Nov 07 15:28:48 piathome librespot[1918]: [2024-11-07T15:28:48Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
Nov 07 15:28:48 piathome librespot[1918]: [2024-11-07T15:28:48Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
Nov 07 15:28:48 piathome librespot[1918]: [2024-11-07T15:28:48Z ERROR librespot_playback::player] PlayerInternal poll: Invalid PlayerState
Nov 07 15:28:48 piathome systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 07 15:28:48 piathome systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 07 15:28:48 piathome systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Nov 07 15:29:00 piathome systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
```

ishiharas, 2024-11-10:
I have a rasperry pi 4 with fresh installed pi os lite on the most recent version. 
I have installed pulseaudio and raspotify. 
When starting raspotify, i am getting the same error in line 3:

```
Nov 10 02:23:37 apollo systemd[1]: Stopped raspotify.service - Raspotify (Spotify Connect Client).
Nov 10 02:23:37 apollo systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Nov 10 02:23:53 apollo librespot[15701]: [2024-11-10T01:23:53Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
Nov 10 02:23:54 apollo librespot[15701]: Failed to create secure directory (/root/.config/pulse): Permission denied
Nov 10 02:23:54 apollo librespot[15701]: Failed to load cookie file from cookie: Permission denied
Nov 10 02:23:54 apollo librespot[15701]: Failed to create secure directory (/root/.config/pulse): Permission denied
Nov 10 02:23:54 apollo librespot[15701]: Failed to load cookie file from cookie: Permission denied
Nov 10 02:23:54 apollo librespot[15701]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Access denied
Nov 10 02:23:54 apollo librespot[15701]: [2024-11-10T01:23:54Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Connection refused (111)'
```

twbonnot, 2024-11-20:
> Hi, maybe I can help but can you post a log which is readable? Right now it's not really useful.

Hey any updates? Could really use some help!

kimtore, 2024-11-20:
Hey, try setting `ProtectHome=false` in `/lib/systemd/system/raspotify.service`, and then reloading systemd and restarting Raspotify?

nilswiersma, 2024-11-22:
Configuring `ProtectHome`:

```
$ cat /lib/systemd/system/raspotify.service | grep ProtectHome
ProtectHome=false
```

Phone client likes raspotify as a device, logging from connecting:

```
$ journalctl -u raspotify -b -f
Nov 22 11:18:08 piathome systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Nov 22 11:19:53 piathome systemd[1]: Stopping raspotify.service - Raspotify (Spotify Connect Client)...
Nov 22 11:19:53 piathome systemd[1]: raspotify.service: Deactivated successfully.
Nov 22 11:19:53 piathome systemd[1]: Stopped raspotify.service - Raspotify (Spotify Connect Client).
Nov 22 11:19:53 piathome systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Nov 22 11:21:07 piathome librespot[1881]: [2024-11-22T11:21:07Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
```

However, trying to play a song twice, different error now:

```
Nov 22 11:24:57 piathome librespot[1881]: [2024-11-22T11:24:57Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
Nov 22 11:25:14 piathome librespot[1881]: [2024-11-22T11:25:14Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
```

nilswiersma, 2024-11-22:
Changing the `LIBRESPOT_DEVICE` device as suggested in https://superuser.com/a/1833079 solved that Audio Sink error for me:

```
cat /etc/raspotify/conf | grep LIBRESPOT_DEVICE=
#LIBRESPOT_DEVICE="default"
LIBRESPOT_DEVICE="hw:CARD=Headphones,DEV=0"
```

Thanks for your help @kimtore !

twbonnot, 2024-11-25:
> Changing the `LIBRESPOT_DEVICE` device as suggested in https://superuser.com/a/1833079 solved that Audio Sink error for me:
> 
> ```
> cat /etc/raspotify/conf | grep LIBRESPOT_DEVICE=
> #LIBRESPOT_DEVICE="default"
> LIBRESPOT_DEVICE="hw:CARD=Headphones,DEV=0"
> ```
> 
> Thanks for your help @kimtore !

Tried this and now I have this error?
/etc/raspotify/conf: Permission denied

This also shows up when I open the file.... Any ideas?

nilswiersma, 2024-11-27:
Try as root?

twbonnot, 2024-11-28:
I honestly don't know what that means.

twbonnot, 2024-12-03:
I figured out my problem. When I downloaded the OS, it was 64bit which was what was recommended, while I actually needed the 32bit legacy.
[image]

closed 2024-12-03 by twbonnot