dtcooper/raspotify #688: Librespot crashes with "Tried too many access points"

kblankenship1989, 2024-11-19:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Raspberry Pi 3b+ running Debian Bullseye.  Just upgraded to latest raspotify and now every time I try to connect from my Spotify account, I get the following error and it fails to connect:
```
ov 18 07:00:26 pandorapi rngd[506]: stats: Lowest ready-buffers level: 2
Nov 18 07:00:26 pandorapi rngd[506]: stats: Entropy starvations: 0
Nov 18 07:00:26 pandorapi rngd[506]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Nov 18 07:05:19 pandorapi systemd[1]: Started Session 41 of user pi.
Nov 18 07:07:02 pandorapi librespot[3698]: [2024-11-18T13:07:02Z WARN  librespot_core::session] Try another access point...
Nov 18 07:07:08 pandorapi librespot[3698]: [2024-11-18T13:07:08Z WARN  librespot_core::session] Try another access point...
Nov 18 07:07:14 pandorapi librespot[3698]: [2024-11-18T13:07:14Z WARN  librespot_core::session] Try another access point...
Nov 18 07:07:21 pandorapi librespot[3698]: [2024-11-18T13:07:21Z WARN  librespot_core::session] Try another access point...
Nov 18 07:07:23 pandorapi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Nov 18 07:07:23 pandorapi systemd[1]: raspotify.service: Succeeded.
Nov 18 07:07:23 pandorapi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Nov 18 07:07:23 pandorapi systemd[1]: Started Raspotify (Spotify Connect Client).
Nov 18 07:08:18 pandorapi librespot[4615]: [2024-11-18T13:08:18Z WARN  librespot_core::session] Try another access point...
Nov 18 07:08:25 pandorapi librespot[4615]: [2024-11-18T13:08:25Z WARN  librespot_core::session] Try another access point...
Nov 18 07:08:31 pandorapi librespot[4615]: [2024-11-18T13:08:31Z WARN  librespot_core::session] Try another access point...
Nov 18 07:08:37 pandorapi librespot[4615]: [2024-11-18T13:08:37Z WARN  librespot_core::session] Try another access point...
Nov 18 07:08:43 pandorapi librespot[4615]: [2024-11-18T13:08:43Z WARN  librespot_core::session] Try another access point...
Nov 18 07:08:50 pandorapi librespot[4615]: [2024-11-18T13:08:50Z ERROR librespot_core::session] Tried too many access points
Nov 18 07:08:50 pandorapi librespot[4615]: [2024-11-18T13:08:50Z ERROR librespot] could not initialize spirc: Deadline expired before operation could complete { timed out }
Nov 18 07:08:50 pandorapi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 18 07:08:50 pandorapi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 18 07:08:50 pandorapi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Nov 18 07:08:50 pandorapi systemd[1]: Starting Raspotify Crash Report Generator...
Nov 18 07:08:51 pandorapi kernel: [112121.571155] pcm512x 1-004d: No SCLK, using BCLK: -2

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
-- System Info --

Linux pandorapi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

-- Logs --

Nov 19 07:46:00 pandorapi librespot[5462]: [2024-11-19T13:46:00Z WARN  librespot_core::session] Try another access point...
Nov 19 07:46:06 pandorapi librespot[5462]: [2024-11-19T13:46:06Z WARN  librespot_core::session] Try another access point...
Nov 19 07:46:13 pandorapi librespot[5462]: [2024-11-19T13:46:13Z WARN  librespot_core::session] Try another access point...
Nov 19 07:46:19 pandorapi librespot[5462]: [2024-11-19T13:46:19Z WARN  librespot_core::session] Try another access point...
Nov 19 07:46:25 pandorapi librespot[5462]: [2024-11-19T13:46:25Z WARN  librespot_core::session] Try another access point...
Nov 19 07:46:31 pandorapi librespot[5462]: [2024-11-19T13:46:31Z ERROR librespot_core::session] Tried too many access points
Nov 19 07:46:31 pandorapi librespot[5462]: [2024-11-19T13:46:31Z ERROR librespot] could not initialize spirc: Deadline expired before operation could complete { timed out }
Nov 19 07:46:31 pandorapi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 19 07:46:31 pandorapi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 19 07:46:31 pandorapi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Nov 19 07:46:31 pandorapi systemd[1]: raspotify.service: Consumed 3.536s CPU time.

-- Config --

LIBRESPOT_QUIET=on
LIBRESPOT_DISABLE_AUDIO_CACHE=on
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=on
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=on
```

kimtore, 2024-11-20:
Hi @kblankenship1989 

Do you have any Spotify-related host overrides in `/etc/hosts`?

Also try disabling the `LIBRESPOT_DISABLE_[...]_CACHE` flags and see if that changes anything?

kblankenship1989, 2024-11-20:
Hosts file:
```
27.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1       pandorapi
```

Config file:
```
# /etc/raspotify/conf -- Arguments/configuration for librespot

# A non-exhaustive list of librespot options and flags.

# Please see https://github.com/dtcooper/raspotify/wiki &
# https://github.com/librespot-org/librespot/wiki/Options
# for configuration details and a full list of options and flags.

# You can also find a full list with `librespot -h`.

# To avoid name collisions environment variables must be prepended with
# `LIBRESPOT_`, so option/flag `foo-bar` becomes `LIBRESPOT_FOO_BAR`.

# Invalid environment variables will be ignored.

# Raspotify defaults may vary from librespot defaults.
# Commenting out the environment variable will fallback to librespot's default
# unless otherwise noted.

# Flags can be set to either "on" or "off".
# Empty values may cause errors.

# Only log warning and error messages.
LIBRESPOT_QUIET=on

# Automatically play similar songs when your music ends.
# Respects client autoplay setting if commented out. Forces autoplay on or off when active.
# This value is either "on" or "off".
#LIBRESPOT_AUTOPLAY=on

# Disable caching of the audio data.
# Enabling audio data caching can take up a lot of space
# if you don't limit the cache size with LIBRESPOT_CACHE_SIZE_LIMIT.
# It can also wear out your Micro SD card. You have been warned. 
# LIBRESPOT_DISABLE_AUDIO_CACHE=on

# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
# LIBRESPOT_DISABLE_CREDENTIAL_CACHE=on

# Play all tracks at approximately the same apparent volume.
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=on

# Enable verbose log output.
#LIBRESPOT_VERBOSE=off

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=off

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
#LIBRESPOT_NAME="Librespot"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
#LIBRESPOT_BITRATE="160"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"

# Displayed device type. Defaults to speaker.
#LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pulseaudio. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""

# Password used to sign in with.
#LIBRESPOT_PASSWORD=""

# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"

# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
#LIBRESPOT_INITIAL_VOLUME="50"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
#LIBRESPOT_VOLUME_CTRL="log"

# Range of the volume control (dB) from 0.0 to 100.0.
# Default for softvol: 60.0.
# For the alsa mixer: what the control supports.
#LIBRESPOT_VOLUME_RANGE="60.0"

# Pregain (dB) applied by volume normalisation from -10.0 to 10.0.
# Defaults to 0.0.
#LIBRESPOT_NORMALISATION_PREGAIN="0.0"

# Threshold (dBFS) at which point the dynamic limiter engages
# to prevent clipping from 0.0 to -10.0.
# Defaults to -2.0.
#LIBRESPOT_NORMALISATION_THRESHOLD="-2.0"

# The port the internal server advertises over zeroconf 1 - 65535.
# Ports <= 1024 may require root privileges.
#LIBRESPOT_ZEROCONF_PORT=""

# HTTP proxy to use when connecting.
#LIBRESPOT_PROXY=""

# ### This is NOT a librespot option or flag. ###
# This modifies the behavior of the Raspotify service.
# If you have issues with this option DO NOT file a bug with librespot.
# 
# By default librespot "download buffers" tracks, meaning that it downloads
# the tracks to disk and plays them from the disk and then deletes them when
# the track is over. This practice is very common, many other audio frameworks
# and players do the exact same thing as a disk based tmp cache is easy to use
# and very resilient. That being said there may be cases where a user may want
# to minimize disk read/writes.
#
# Commenting this out will cause librespot to use a tmpfs so that provided there
# is enough RAM to hold the track nothing is written to disk but instead to a tmpfs.
# See https://github.com/dtcooper/raspotify/discussions/567
# And https://www.kernel.org/doc/html/latest/filesystems/tmpfs.html
TMPDIR=/tmp
```
Same error

kblankenship1989, 2024-11-21:
For the record, I have tried running `librespot` on its own and confirmed this is an issue there.  I have raised [THIS ISSUE](https://github.com/librespot-org/librespot/issues/1401) with them to address this

kblankenship1989, 2024-11-21:
until this is resolved for me, is there any way to downgrade raspotify (I cannot figure out how to install an older version if that is even possible, still fairly new to linux)

kblankenship1989, 2024-11-25:
So here is where I am @kimtore:
- Tried all the suggestions above, same results
- Tried running librespot directly on my pi, same issue
- Tried running librespot on my Linux PC, no issues (narrowed it to my pi), so I closed the issue with librespot as it is not an issue with them (they don't necessarily claim to support Raspberry Pi's to my knowledge)
- Tried brand new SD card, brand new install of raspotify following the documentation with the install script and all,  No changes to config file, same issue
- Tried on a Pi 4 I have (I have been using a Pi 3b+ for the last 5 yrs now with raspotify), same issue
So not sure what the changes are or how they are causing issues but the changes in librespot appear to be causing issues at least for my raspberry pi's with respect to connecting to access points.

kblankenship1989, 2024-11-25:
SOLVED!!!!!!!!

For anyone else that runs into this, the root cause was my `NetworkManager` on my pi was configured to point to `192.168.1.104` as a primary DNS on my network, with `192.168.1.1` as a backup.  Now why is this an issue, you ask? BECAUSE '192.168.1.104` DOES NOT EXIST ON MY NETWORK!!!!!  Things like pinging `google.com` would still resolve, only delayed because it would attempt the `.104` address first, then fail and finally use the `.1` address and work, but librespot closes its connection before the pi has a chance to attempt the fallback DNS.  Once I followed [THESE DIRECTIONS](https://www.reddit.com/r/raspberry_pi/comments/18t6joj/dns_on_a_headless_rpi_os_not_resolving/?rdt=56138#:~:text=To%20change%20DNS%20settings%20using,show%20to%20list%20available%20connections.&text=Replace%20%5Bconnection_name%5D%20with%20your%20connection's,your%20desired%20DNS%20server%20address.) to reset my DNS to `192.168.1.1` as primary, it all worked like magic!!!

closed 2024-11-25 by kblankenship1989

onefrostyone, 2025-01-09:
@kblankenship1989 thank you so much! Had nearly the same issue and this helped immensely! 

Fwiw I had messed around with router ip addresses and the raspberry pi was trying to run through a pihole dns server. I think this was causing timeout issues. So. Not sure exactly what was the problem but changing the DNS to go directly through the router fixed it.