dtcooper/raspotify #689: No spotify connect after first reboot

Coumodo, 2024-12-09:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

instant after installation on Zero 2 W 32bit lite OS raspotify is working well, but only until first reboot.
After reboot Spotify Connect doesn't show up any more on all other devices in same wifi.

I assume issue with librespot due respotify service log

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
○ raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; preset: enabled)
     Active: inactive (dead) since Mon 2024-12-09 18:55:08 CET; 10s ago
   Duration: 288ms
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 1308 ExecStart=/usr/bin/librespot (code=exited, status=1/FAILURE)
   Main PID: 1308 (code=exited, status=1/FAILURE)
        CPU: 289ms

Dec 09 18:55:08 AudiPi systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Dec 09 18:55:08 AudiPi librespot[1308]: [2024-12-09T17:55:08Z ERROR librespot] Invalid `--autoplay` / `-A`: ""
Dec 09 18:55:08 AudiPi librespot[1308]: Valid `--autoplay` / `-A` values: on, off
Dec 09 18:55:08 AudiPi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Dec 09 18:55:08 AudiPi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dec 09 18:55:08 AudiPi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
```

kimtore, 2024-12-13:
Please check the value of `LIBRESPOT_AUTOPLAY` in `/etc/raspotify/conf` and verify that it is either `ON` or `OFF`, no other values are allowed.

Coumodo, 2024-12-13:
> Please check the value of `LIBRESPOT_AUTOPLAY` in `/etc/raspotify/conf` and verify that it is either `on` or `off`, no other values are allowed.

thanks, correct hint by you, Issue solved adding LIBRESPOT_AUTOPLAY=**_"on"_** in conf file.
Strange that this stetting marker is left empty by default with installation?!

closed 2024-12-13 by Coumodo

kimtore, 2024-12-13:
> Strange that this setting marker is left empty by default with installation?!

These values were previously allowed to be empty, but broke with the upgrade to librespot 0.5.0. See commit b927945884b92900ba7079d2784d7a1998395e1d.

Re-installing or upgrading will not replace your configuration file, but prompt you whether you want to keep your existing one or install the package maintainer's version. Unfortunately this leads to breakage. Happy to hear the fix worked for you.