dtcooper/raspotify #692: Autoplay doesn't work

kramer65, 2024-12-23:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

When I play one song on spotify, raspotify stops playing afterwards, instead of playing similar songs. 

I'm on raspotify the latest version (0.45.0):
```
$ apt search raspotify
Sorting... Done
Full Text Search... Done
raspotify/unknown,now 0.45.0~librespot.v0.5.0-84d28e8 arm64 [installed]
  Raspotify
```

I enabled autoplay in the config:
```
$ sudo grep AUTOPLAY /etc/raspotify/conf 
LIBRESPOT_AUTOPLAY=on
```

I also found [this issue](https://github.com/dtcooper/raspotify/issues/430), in which @dtcooper suggests to use the `--autoplay` flag of librespot. But I'm unaware how I should use that flag. I thought that setting LIBRESPOT_AUTOPLAY to on would set the librespot `--autoplay` flag, but it seems it doesn't. Do I simply change the in `ExecStart` in `/lib/systemd/system/raspotify.service` to something like `ExecStart=/usr/bin/librespot --autoplay`? Or is there a different place where I should insert the autoplay flag?

Note that I've included some logs, although I don't think that reveals any errors relating to the autoplay thing. 

All tips are welcome!

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
$ sudo journalctl -u raspotify -b | grep "Dec 23"
Dec 23 08:26:04 volumio librespot[8595]: [2024-12-23T07:26:04Z ERROR librespot_connect::spirc] ContextError: Error { kind: NotFound, error: StatusCode(404) }
Dec 23 08:26:18 volumio librespot[8595]: [2024-12-23T07:26:18Z ERROR librespot_connect::spirc] ContextError: Error { kind: NotFound, error: StatusCode(404) }
Dec 23 08:26:31 volumio librespot[8595]: [2024-12-23T07:26:31Z ERROR librespot_connect::spirc] ContextError: Error { kind: InvalidArgument, error: StatusCode(400) }
Dec 23 08:26:33 volumio librespot[8595]: [2024-12-23T07:26:33Z WARN  librespot_playback::player] This track may exceed dBFS by 9.01 dB and be subject to 11.01 dB of dynamic limiting at its peak.
Dec 23 08:27:26 volumio librespot[8595]: [2024-12-23T07:27:26Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'Broken pipe (32)'
Dec 23 08:27:26 volumio librespot[8595]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Dec 23 08:27:42 volumio librespot[8595]: [2024-12-23T07:27:42Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
Dec 23 11:15:46 volumio librespot[8595]: [2024-12-23T10:15:46Z ERROR librespot_connect::spirc] ContextError: Error { kind: InvalidArgument, error: StatusCode(400) }
Dec 23 11:15:47 volumio librespot[8595]: [2024-12-23T10:15:47Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'Broken pipe (32)'
Dec 23 11:15:47 volumio librespot[8595]: ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred
Dec 23 11:16:52 volumio librespot[8595]: [2024-12-23T10:16:52Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
Dec 23 11:16:55 volumio librespot[8595]: [2024-12-23T10:16:55Z ERROR librespot_connect::spirc] ContextError: Error { kind: InvalidArgument, error: StatusCode(400) }
Dec 23 11:16:57 volumio librespot[8595]: [2024-12-23T10:16:57Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
[... 141 lines cut ...]

-- Ouput of aplay -L --

null
    Discard all samples (playback) or generate zero samples (capture)
default
hw:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=IQaudIODAC
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Default Audio Device
dmix:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Direct sample mixing device

-- Ouput of librespot -d ? --

	Compatible alsa device(s):

	------------------------------------------------------
```

kimtore, 2024-12-27:
Hi @kramer65,
Actually `LIBRESPOT_AUTOPLAY` will have the exact same result as setting `--autoplay`, I confirmed this in the librespot source code. If you want to confirm then you can edit the systemd file like you suggest. But I don't think you'll have any luck.

To start, I suggest first updating to version Raspotify 0.46. Perhaps this behavior stems from a bug/protocol issue and is fixed in the latter version of librespot. Not sure.

Omitting the autoplay option will let the client control the autoplay. If you are you using an official client this will usually work just fine, probably for other clients too but I cannot tell. Perhaps your client has an option for this.

Let me know and I'll try to help you further.

kimtore, 2024-12-27:
As a side note you can also try running with `LIBRESPOT_VERBOSE=on` instead of quiet, this will produce more log that might shed some light on the problem.

kramer65, 2024-12-29:
Since a couple days the issue seems to be solved. I just noticed that it autoupgraded to version 0.46 and I guess that solved the issue for me. Thanks for creating raspotify! I'm using it for many hours every day now, and on new years eve it will power my home party! 🥳

closed 2024-12-29 by kramer65

kimtore, 2025-01-02:
Great stuff, glad you got it working and hope your party went well!

Side note: I can't take credit for creating librespot or raspotify (cc: @dtcooper), but I will humbly accept credit for keeping it somewhat up-to-date ;)