dtcooper/raspotify #704: Impossibile (for me) to use ALSA MIXER  with Raspotify

lula73, 2025-01-20:
### Compatible Device

- [x] I'm running Raspotify on a compatible Device

### Are you sure?

- [x] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [x] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [x] I'm running Raspotify on a compatible Configuration

### Latest Version

- [x] I'm running the latest version of Raspotify

### Are you sure?

- [x] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [x] I have done my due diligence

### What happened?

I apologize for having to bother you, but after 10 days of attempts... I'm starting to think there might be a bug or maybe a problem in the documentation. 

I'm building a voice assistant application for Raspberry Pi 5 and I've integrated Spotify with Raspotify.
Raspotify/Librespot works perfectly and the music works, but I can't share the audio device with other applications.

I removed pulseAudio and set up only ALSA with Mixer, and when I use two concurrent programs (e.g., VLC and Browser), the music mixes perfectly.

But this doesn't happen with Raspotify: if a playback starts and music is playing, other applications can't access the audio.
If another application like vlc starts for first, then Raspotify errors out and the device isn't found or busy.

```
Connection Refused: <AlsaSink> Device dmixer May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Device or resource busy (16)'
gen 20 22:09:18 raspberrypi librespot[47900]: [2025-01-20T21:09:18Z ERROR librespot_playback::player] PlayerInternal poll: Invalid PlayerState

```

Since asound.conf and ALSA mixer work at the OS level because the mixing works,  the problem is in the Raspotify CONF, where, regardless of the dozens and dozens of attempts made, it seems to take the entire device without passing through the mixer. I believe mixer management for Raspotify is essential, and I haven't found clear documentation on this.

I hope someone can help me because mixing is essential for applications, and anyway, the operating system sounds must coexist with the music. Tnks! Luca

An Example of my raspotify Conf
```
LIBRESPOT_NAME="BMO"
LIBRESPOT_BACKEND="alsa"
LIBRESPOT_DEVICE="dmixer"
OPTIONS="--mixer alsa"
LIBRESPOT_VOLUME_CONTROL="fixed"
```
Others variations

```
#OPTIONS="--device plug:default --device-type alsa"
#BACKEND_ARGS="--device plug:default"
#OPTIONS="--backend alsa --device dmixer --mixer alsa  --linear-volume 
#BACKEND_ARGS="--device dmixer"
#OPTIONS="--device dmixer --device-type alsa --mixer alsa --alsa-mixer-control Master --alsa-mixer-device dmixer"
#OPTIONS="--device dmixer --device-type alsa --mixer softvol"
```

My asound.conf in attach

[asound.txt](https://github.com/user-attachments/files/18482548/asound.txt)

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Connection Refused: <AlsaSink> Device dmixer May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Device or resource busy (16)'
gen 20 22:09:18 raspberrypi librespot[47900]: [2025-01-20T21:09:18Z ERROR librespot_playback::player] PlayerInternal poll: Invalid PlayerState
```

kimtore, 2025-01-20:
Hi @lula73,

Sorry about your debugging issues, I'll try to help you, it's no bother.

I can think of two possible things that's going wrong for you:
1. Perhaps it will help to give `librespot` the dmix device, on my system this is `dmix:CARD=sndrpihifiberry,DEV=0` and you can find it with `aplay -L | grep dmix`. This way I suppose the device will not be locked.
2. Alternatively, maybe there are some user/permission problems, however I think librespot is running as root with some constrained permissions; you can have a look in `/lib/systemd/system/raspotify.service` and tweak some parameters there?

kimtore, 2025-01-20:
Also as a side note, the configuration options should work fine, they are read as-is by Librespot and you can see all the options in the [librespot source code](https://github.com/librespot-org/librespot/blob/b54f3e30db99cb665ac054e9860a2b29c21bca43/src/main.rs#L231-L280). There is also a [wiki with options](https://github.com/librespot-org/librespot/wiki/Options).

kimtore, 2025-01-20:
Perhaps another suggestion is to try running Librespot manually from the command line and tweak the options? This way you can rule out any misconfiguration issues from Raspotify.

lula73, 2025-01-21:
Thank you for the suggestions, i will try them  tonight  one by one. Tnks!

lula73, 2025-01-21:
The problem seem finally resolved.  It's related to permission 

edit and change 

`sudo nano /lib/systemd/system/raspotify.service`

```
[Service]

RemoveIPC=false             # changed from true to false
DevicePolicy=auto          # change from strict to auto

User=bmo                    #  insert user that run raspotify
Group=audio                 # the usergroup group
SupplementaryGroups=audio pulse-access

```

and finally : 

```
sudo systemctl daemon-reload
sudo systemctl restart raspotify

```
 
Hope other users will not loose 10 days... : D

Tnks for indications  Kim!!!

kimtore, 2025-01-23:
Very nice! Glad you got it working.

Upgrading Raspotify will reset your changes. To make your changes permanent across upgrades you should edit the service through systemctl.
```systemctl edit raspotify```

I will consider making some changes to the default configuration so that ALSA device sharing can work out of the box, your debugging results will be helpful in this regard. It will require some more testing on my part though. And perhaps Raspotify should have its own user with well-defined permissions.

lula73, 2025-01-24:
It seems incredible, but this morning I tested and running VLC music work perfectly mixing  with raspotify music. Never made reboot in the last days,  But After a reboot, the mixing  doesn't work anymore. 
And I can't get it to work again...
All configurations are the same. It's really complex to manage in this way. 
I noticed that I still have pulseaudio in the OS, try to unininstall but  after install again.  

So i get again the same errors when i try to run spotify with VLC running : 

```
gen 24 18:16:44 raspberrypi librespot[3939]: [2025-01-24T17:16:44Z ERROR librespot_connect::spirc] ContextError: Error { kind: NotFound, error: StatusCode(404) }
gen 24 18:16:44 raspberrypi librespot[3939]: Home directory not accessible: Permission denied
gen 24 18:16:44 raspberrypi librespot[3939]: ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
gen 24 18:16:44 raspberrypi librespot[3939]: [2025-01-24T17:16:44Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
gen 24 18:16:44 raspberrypi librespot[3939]: ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
gen 24 18:16:44 raspberrypi librespot[3939]: [2025-01-24T17:16:44Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'Unknown errno (524)'
gen 24 18:16:44 raspberrypi librespot[3939]: [2025-01-24T17:16:44Z ERROR librespot_playback::player] PlayerInternal poll: Invalid PlayerState
gen 24 18:16:44 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
gen 24 18:16:44 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
gen 24 18:16:44 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
```
`

closed 2025-04-25 by kimtore