dtcooper/raspotify #705: Get premium to listen

ZM71, 2025-01-25:
### Compatible Device

- [x] I'm running Raspotify on a compatible Device

### Are you sure?

- [x] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [x] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [x] I'm running Raspotify on a compatible Configuration

### Latest Version

- [x] I'm running the latest version of Raspotify

### Are you sure?

- [x] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [x] I have done my due diligence

### What happened?

Get premium to listen on this device and my tv box and Chromecast work fine. https://i.ibb.co/V3rHJNq/Screenshot-2025-01-25-20-08-02-467-com-spotify-music.jpg

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
None because the app is working 😡
```

ZM71, 2025-01-25:
What's wrong?

ZM71, 2025-01-27:
Why make the app to require premium?

dtcooper, 2025-01-27:
Are you using a comparable OS? 

Your screenshot indicates you are not (DietPi)

DietPi may not have zeroconf per the troubleshooting page on the wiki and you may need to enter your credentials.

closed 2025-01-27 by github-actions[bot]

ZM71, 2025-01-27:
What the actual fuck?

ZM71, 2025-01-27:
> Are you using a comparable OS? 
> 
> Your screenshot indicates you are not (DietPi)
> 
> DietPi may not have zeroconf per the troubleshooting page on the wiki and you may need to enter your credentials. 
It's it but it is with xfce4. Bruh

ZM71, 2025-01-27:
> Are you using a comparable OS? 
> 
> Your screenshot indicates you are not (DietPi)
> 
> DietPi may not have zeroconf per the troubleshooting page on the wiki and you may need to enter your credentials. 
I don't understand how to use this dumb app.

dtcooper, 2025-01-27:
Please follow the instructions on the wiki, beyond that we don't offer troubleshooting support.