dtcooper/raspotify #706: Can't see device - troubleshooting help?

DavidRGriswold, 2025-01-26:
### Due Diligence

- [x] I have done my due diligence

### What can we help you with?

I just installed raspotify on my raspberry pi 4, and it simply is not appearing on my network. The pi is connected over Ethernet. I have tried clients connected over wifi and also my PC connected to the same router as the pi, nothing show up. Nothing shows up in journalctl except start and stop notices. No crash report. Service shows as running. 

I use multiple google homes which show up no problem. I have also used an Android device as a Spotify connect receiver with an APK I found online, which also uses librespot as a backend,  and  it appears fine (though it doesn't connect, but that's a different issue).

Is there a guide or series of steps I can try to follow to pinpoint *where* the issue is occuring? I'm guessing it has something to do with zeroconf over my router but I am in a bit over my head. I'm very tech capable just not hugely knowledgeable about network stuff.

Panp59, 2025-01-27:
quite same problem here, all my devices can be managed by some headless apps I have, but raspotify can't, I must start my spotify app (on my phone for example), and next I can. Is it a way to manually "wake-up" it ?

doris1347, 2025-01-28:
Did you do OAuth?

doris1347, 2025-01-28:
Okay, a quick note for users who are feeling confused about this:
FYI dtcooper/raspotify#675

OAuth: Authorization

Password login has been deprecated and is no longer supported. Instead, OAuth Authorization must be used. Follow the steps below at START HERE to authenticate using OAuth:

To use Midori browser under vnc comment out #python3 spotifypod.py in sudo nano /etc/xdg/openbox/autostart
and save and reboot ...right click on the black openbox screen then open web browser (will take a while to load) Open the provided URL to complete the login process, you should see a hyperlink that says login to spotify, click it, you will be redirected to spotify login, just login and accept the terms then:-

[INFO librespot_oauth] OAuth server listening on 127.0.0.1:5588

The authentication token will be in the folder YOUR_CONFIGURATION_PATH called credentials.json. I copied that now to the raspotify folder cp credentials.json /var/cache/raspotify/

Updated the raspotify config file sudo nano /etc/raspotify/conf to comment out below items to enable 0Auth instead of username/password:

#LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
#LIBRESPOT_DISABLE_DISCOVERY=
#LIBRESPOT_USERNAME="mysecretusername"
#LIBRESPOT_PASSWORD="mysecretpassword"

Restarted raspotify with systemctl restart raspotify and everything worked,

In sudo nano /etc/xdg/openbox/autostart un-comment python3 spotifypod.py save and reboot.

Good Luck.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Dec 31 00:07:16 2024

START HERE:

Password login has been deprecated and is no longer supported. Instead, OAuth Authorization must be used. Follow the steps below to authenticate using OAuth:

#Run the following command to initiate the OAuth process:

librespot --cache /var/cache/raspotify -j
Open the provided URL in a web browser (Midori) to complete the login process

Press Ctrl-C to quit. [2024-12-31T05:17:30Z INFO librespot] Gracefully shutting down.

pi@sPoT:$ cd /var/cache/raspotify/
-bash: cd: /var/cache/raspotify: Permission denied
pi@sPoT:$ sudo chmod -R a+rwX /var/cache/raspotify To obtain read/write permission to /var/cache/raspotify directory.

pi@sPoT:$ cd /var/cache/raspotify
pi@sPoT:/var/cache/raspotify$ ls -a
. .. credentials.json files volume
pi@sPoT:/var/cache/raspotify$ cd
pi@sPoT:$ systemctl daemon-reload && systemctl restart raspotify

==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ===
Authentication is required to reload the systemd state.
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ===
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'raspotify.service'.
Authenticating as: ,,, (pi)
Password:

Update: the raspotify config file sudo nano /etc/raspotify/conf to comment out below items to enable 0Auth instead of username/password:

#LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
#LIBRESPOT_DISABLE_DISCOVERY=
#LIBRESPOT_USERNAME="mysecretusername"
#LIBRESPOT_PASSWORD="mysecretpassword"

sudo reboot

note: However you may still need to select the device from the spotify-connect list (if you didn't do OAuth (Open Authorization) with your computer/phone first you may have to refresh the Spotify-Connect list a few times to be able to see your "sPoT" device on the list of available devices - you can close immediately after otherwise you will receive "error no devices".

DavidGriswoldTeacher, 2025-01-28:
In my case, I wasn't authenticating, just trying to use Zeroconf. The problem in my case was that I was connected to a VPN (NordVPN) that enabled a firewall. I disabled the firewall and things were fine. This can definitely be closed, though that is a helpful post!

closed 2025-05-06 by github-actions[bot]