dtcooper/raspotify #713: Documentation doesn't reflect librespot update

mruijzendaal, 2025-02-10:
### Raspotify is just a package

- [x] I know Raspotify is just a package

### Due Diligence

- [x] I have done my due diligence

### What can we do to make Raspotify more awesome?

Release https://github.com/dtcooper/raspotify/releases/tag/0.44.1 removes the `LIBRESPOT_SAMPLE_RATE` and `LIBRESPOT_INTERPOLATION_QUALITY` flags, but this is not yet reflected in the [wiki](https://github.com/dtcooper/raspotify/wiki/Configuration). Similarly, the new configuration style (on/off) is not yet implemented.

I was working with a 48kHz subsample rate, which now appears to be incompatible with raspotify/librespot. Could any update in documentation help my case?

dtcooper, 2025-02-14:
@kimtore something you could look at?

kimtore, 2025-02-18:
Yes, however, I am going to be quite busy this week so would have to wait a while.

BigFood2307, 2025-02-20:
I am a total noob when it comes to audio stuff in Linux, but I had pretty much the same issue.
After a lot of searching, i found this:
#331 
I had to use speexrate_medium instead of lavrate_higher, otherwise I would get stuttering. Also remember to exchange CODEC for your own device name.
Then simply set the Raspotify device to default, and restart raspotify. Now I can listen on my crappy DAC, that only has 48kHz.
Not sure this is the best solution, but it might help you for now.

kimtore, 2025-08-13:
@mruijzendaal one can only get 44.1kHz sample rate from librespot now. Resampling is not supported anymore. Which means your options are pretty much to have ALSA or Pipewire do this for you.

I did update the default configuration file, it is reflected in the Wiki, perhaps you could use something there?

mruijzendaal, 2025-08-14:
Thanks for updating the documentation! After diving deeper into librespot and alsa I managed to get it working again. I would have liked to share it here, though I'm afraid I'm not sure what specific change fixed my issues.

closed 2025-08-14 by mruijzendaal