dtcooper/raspotify #721: Installed perfectly worked since last night

sonammrasp, 2025-05-13:
### Due Diligence

- [x] I have done my due diligence

### What can we help you with?

Hello Sir,

I hope you're doing well. I installed Raspotify two days ago, and everything was working fine. However, today when I tried to switch the device from Spotify on my mobile, it wouldn’t switch. After attempting two or three times, the song paused, and then Raspotify disappeared from the device list for a few minutes. It reappeared afterward, but I am unable to play music on the Raspberry Pi anymore.

Could you please assist me in resolving this issue?

marpisco, 2025-05-20:
Attempt restarting it.

kimtore, 2025-05-21:
If restarting didn't help, I'm prepared to have a look, but would need some logs from librespot as well.

closed 2025-08-09 by kimtore