dtcooper/raspotify #729: `apt` is not seeing the latest build

daweimau, 2025-08-12:
### Due Diligence

- [x] I have done my due diligence

### What can we help you with?

Please forgive a linux noobie. 

```bash
 $ sudo apt update
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 https://dtcooper.github.io/raspotify raspotify InRelease
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:4 http://deb.debian.org/debian bullseye InRelease
Hit:5 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
328 packages can be upgraded. Run 'apt list --upgradable' to see them.

$ sudo apt install --only-upgrade raspotify
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspotify is already the newest version (0.46.1~librespot.v0.6.0-383a6f6).
```

From all I can see, the the 0.46.2 release should be now available. Am I missing something?

codydubat, 2025-08-12:
I believe this action needs to be triggered manually: 
https://github.com/dtcooper/raspotify/actions/workflows/apt-deploy.yml
Is this correct @kimtore  ? 
If so, could some trigger be set to run this automatically on a new release?

codydubat, 2025-08-12:
FYI, if you don't want to wait until it's available in apt, installing the deb manually via the RPi desktop environment should work just fine. (it did for me)

MichaIng, 2025-08-12:
Right. It could be triggered on creating a tag which follows a specific pattern:
```yml
on:
  create:
  workflow_dispatch:
    ...

jobs:
  apt_deploy:
     if: ${{ github.event_name == 'workflow_dispatch' || github.ref =~ '^refs/tags/[0-9]+\.[0-9]+\.[0-9]+$' }}
     ...
```
So it remains possible to trigger it manually, but it also runs on `create` event, if triggered from creating a tag which matches a semantic version string. Just as an idea 🙂. Hoping for seeing those soon in the APT repo, since the new builds support Debian Trixie for 32-bit systems. Currently, the `libasound2` dependency (not allowing `libasound2t64`) breaks it.

kimtore, 2025-08-13:
Thanks for your patience, and in pointing out improvements to the workflow.

Package should be in the repos now.