dtcooper/raspotify #736: Raspotify install fails on Ubuntu 25.04 (arm64) due to libasound2 virtual package (t64 transition)

TheRoyalGhost45, 2025-08-13:
### Compatible Device

- [x] I'm running Raspotify on a compatible Device

### Are you sure?

- [x] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [x] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [x] I'm running Raspotify on a compatible Configuration

### Latest Version

- [x] I'm running the latest version of Raspotify

### Are you sure?

- [x] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [x] I have done my due diligence

### What happened?

On a Raspberry Pi 4 running Ubuntu 25.04 (arm64), the official install script fails when resolving dependencies because libasound2 is now a virtual package and needs to be explicitly satisfied by libasound2t64 on recent Ubuntu releases, and the installation still fails even after installing libasound2t64 manually with sudo apt install libasound2t64.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
@pihome:~$ sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
[sudo] password for :
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (8.12.1-3ubuntu1).
Solving dependencies... Done
The following packages were automatically installed and are no longer required:
  libavahi-client3 libavahi-common-data libavahi-common3 libavahi-core7 libdaemon0 libnss-mdns
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Hit:1 http://ports.ubuntu.com/ubuntu-ports plucky InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports plucky-updates InRelease [126 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports plucky-backports InRelease [126 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports plucky-security InRelease [126 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports plucky-updates/main arm64 Packages [286 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports plucky-updates/main arm64 Components [27.4 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports plucky-updates/universe arm64 Packages [183 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports plucky-updates/universe arm64 Components [11.3 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports plucky-updates/restricted arm64 Components [212 B]
Get:10 http://ports.ubuntu.com/ubuntu-ports plucky-updates/multiverse arm64 Components [212 B]
Get:11 http://ports.ubuntu.com/ubuntu-ports plucky-backports/main arm64 Components [212 B]
Get:12 http://ports.ubuntu.com/ubuntu-ports plucky-backports/universe arm64 Components [212 B]
Get:13 http://ports.ubuntu.com/ubuntu-ports plucky-backports/restricted arm64 Components [216 B]
Get:14 http://ports.ubuntu.com/ubuntu-ports plucky-backports/multiverse arm64 Components [216 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports plucky-security/main arm64 Packages [195 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports plucky-security/main arm64 Components [14.7 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports plucky-security/universe arm64 Packages [106 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports plucky-security/universe Translation-en [37.2 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports plucky-security/universe arm64 Components [1660 B]
Get:20 http://ports.ubuntu.com/ubuntu-ports plucky-security/restricted arm64 Components [212 B]
Get:21 http://ports.ubuntu.com/ubuntu-ports plucky-security/multiverse arm64 Components [212 B]
Fetched 1241 kB in 2s (629 kB/s)
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Package libasound2 is a virtual package provided by:
  libasound2t64 1.2.13-1ubuntu0.1 (= 1.2.13-1ubuntu0.1)
  liboss4-salsa-asound2 4.2-build2020-1ubuntu3
You should explicitly select one to install.

Error: Package 'libasound2' has no installation candidate
```

bycrea, 2025-10-18:
Worked for me installing with .deb package [raspotify-latest_armhf.deb](https://dtcooper.github.io/raspotify/raspotify-latest_armhf.deb)
Not with curl && install.sh

Thanks for your work !

kimtore, 2025-11-07:
Hi, please try the new version 0.47.1 and see if it it is fixed yet?

MichaIng, 2025-11-07:
Ohh, actually, the issue is not the Raspotify package, but the installer, which tries to install `libasound2`. So the dependency declaration was most likely the wrong track: https://github.com/dtcooper/raspotify/blob/master/install.sh#L19C1-L19C13

One way would be to check whether `libasoundt64` is available, and install that one explicitly in case. But easier is to just drop the `libasound2` package from `REQ_PACKAGES`. It is a dependency of `alsa-utils` anyway, and the installer declares the same minimum version for both. So no point to check and install explicitly `libasound2` or any of its variants.

MichaIng, 2025-11-07:
This should fix it: #750

```console
root@VM-Forky:~#  curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Hit:1 https://deb.debian.org/debian forky InRelease
Hit:2 https://deb.debian.org/debian forky-updates InRelease
Hit:3 https://deb.debian.org/debian-security forky-security InRelease
Hit:4 https://deb.debian.org/debian forky-backports InRelease
Hit:5 https://dietpi.com/apt forky InRelease
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Notice: Some sources can be modernized. Run 'apt modernize-sources' to do so.
Package libasound2 is a virtual package provided by:
  liboss4-salsa-asound2 4.2-build2020-6
  libasound2t64 1.2.14-2 (= 1.2.14-2)
You should explicitly select one to install.

Error: Package 'libasound2' has no installation candidate
root@VM-Forky:~#  curl -sL https://raw.githubusercontent.com/MichaIng/raspotify/refs/heads/master/install.sh | sh

...

Installing:
  alsa-utils  libpulse0

Installing dependencies:
  libasound2-data  libasound2t64  libasyncns0  libatopology2t64  libfftw3-single3  libflac14  libgomp1  libmp3lame0  libmpg123-0t64  libogg0  libopus0  libsamplerate0  libsndfile1  libvorbis0a  libvorbisenc2  libx11-6  libx11-data  libx11-xcb1  libxau6  libxcb1  libxdmcp6

...

Installing:
  raspotify

Installing dependencies:
  avahi-daemon  libavahi-common-data  libavahi-common3  libavahi-core7  libdaemon0

...

Thanks for installing Raspotify! Don't forget to checkout the wiki for tips, tricks and configuration info!:

https://github.com/dtcooper/raspotify/wiki

And if you're feeling generous you could buy me a RedBull:

https://github.com/sponsors/JasonLG1979
```

But as mentioned in the PR as alternative solution: why not skipping this manual dependency handling, and let the APT package do this?

kimtore, 2025-11-10:
> This should fix it: [#750](https://github.com/dtcooper/raspotify/pull/750)

Thanks, I merged it.

> But as mentioned in the PR as alternative solution: why not skipping this manual dependency handling, and let the APT package do this?

Yes, interesting idea, any idea why this is handled manually?

kimtore, 2025-11-12:
> Maybe at first, the DEB package did not declare dependencies yet, and while they were added later on, they were kept in the installer regardless. I mean it does cause issues (until now) to install them manually beforehand.

Thanks for digging in the history. Yes, that sounds likely.

> But indeed could be a cleaned up and reduce maintenance to remove this. Also has the benefit that after uninstall, `apt autoremove` can remove Raspotify dependencies not needed for other (manually installed) DEB packages.

Indeed! Less code equals less potential problems. I created #752 for this purpose. Let's continue the discussion there and close this ticket.

closed 2025-11-12 by kimtore